iNavFlight/inav #2906: HOTT Telemetry

Basti77, 2018-03-11:
Deutsch - erkannt
Englisch
Is hott telemetry no longer supported?

I tried it on the F35 and for the test on a Naze. I can not get it to work. Try with and diode.
For info: betaflight does not require a diode anymore.
 INaV 1.73

Ralfde, 2018-03-11:
inav doesn't need a diode any more, too. But use inav 1.8.0 or 1.9.0.
Some F3 boards doesn't support HoTT any more, because the memory is not big enough and is used f.e. for the OSD function.
You should try the Omnibus F4 SD V3. This works good with inav and HoTT.

Basti77, 2018-03-11:
I have one F35 Lightning Flightcontroller here.
INAV 1.9.0, no telemetry on any UART.

So the article is not correct anymore:
https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md

Ralfde, 2018-03-11:
As I know it can be connected like this, if you switch the half dublex mode off. But now it can be connected also without diode direct to any TX pin, because it's now already the half dublex mode standard. So just connect the telemetry cable to any TX pin or the softserial.

Basti77, 2018-03-12:
meanwhile I believe that the F35 was delivered broken.
also at the VTX output no voltage comes out.

Basti77, 2018-03-12:
I'm reading the target.h files of the boards.
at the Naze Target you can see the line commented out.
// # define USE_TELEMETRY_HOTT
you called the reason.

FF_F35_LIGHTNING has nothing at all.
OMNIBUSF4 is not synonymous, but you say that it works there.
How can I see that HoTT is active in the code for the f35 board?

shellixyz, 2018-03-12:
HOTT telemetry is not available (not enabled) in the release builds for Naze because of the lacking available flash size but it is enabled for all F3 and F4 targets including FF_F35_LIGHTNING: https://github.com/iNavFlight/inav/blob/master/src/main/target/common.h#L89

Basti77, 2018-03-12:
I found a mistake on the board.
Regardless of the Hott story, there is no voltage on the VTX Port.

I opened the case and an IC is defective. (The FC was brand new and not yet installed)
you see with your naked eye.

The dealer changes the device and then we continue...

Basti77, 2018-03-24:
Yesterday I got a new FC.
There is VTX now working.
INAV has been updated.
But still no Hott telemetry.
I've tried all Uarts, other RX.
There is nothing.

Does anyone
[image]
 have an idea?

closed 2018-06-06 by stale[bot]