iNavFlight/inav #2907: when switch to horizon mode , one servo moves

bloktar, 2018-03-11:
hi , mi board , omnibus f3 v2. inav 1.8 on a s800 flying wing.
when switch to horizon mode one servo moves a little but on the air this little makes the wing go up.

the board is placed neutral and all seems to work like this problem that i dont know what happen.

shellixyz, 2018-03-11:
Sounds like your board pitch alignment or your accelerometer calibration is not right. When your wing is flat relative to the ground if the roll and pitch angles shown in the setup tab are 0 it shouldn't do that. It should go down (angle of attack = 0). Then you need to adjust the pitch alignment to the desired angle of attack to make the wing fly level. If it is too high your wing will raise.

bloktar, 2018-03-11:
the board i think is nice niveled becouse its on the top of a printed part with stands.. but i was moving the wing meanwhile switching manual mode to horizon and on 5 positive pitch the servo stop moving. i put 5 positive on pitch on the configurator and now on the same position where the wing moved the servo now the servo dont move when switch to horizon. i think its solved. thanks for your  fast answer and help shellixyz. and sorry for my english xD i hope you understand.

shellixyz, 2018-03-11:
Nice :+1: ! If your issue is solved please close this report.

closed 2018-03-11 by bloktar