iNavFlight/inav #2908: the on inav 1.9.0

Dhananjaya96, 2018-03-12:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

Dhananjaya96, 2018-03-12:
I am running with AnyFC F7 .and after upgrading to inav 1.9.0 external baro the mag declination is not working .
what I tried . 1 after calibrating compass I changed the mag declination and clicked on save and reboot but its not working .same thing I tried on inav 1.7.3 external baro its working perfectly.

then comes to blackbox .its a cool feature available on 1.9.0. but it also have problem .when you power up and armed your craft no data recorded and showing fatal error 

if you connect it with pc and click on save reboot the after that it works . meaning once power up then connect it with pc click on save and reboot then its work. but in beta flight no need of this once power up black box is ready to do log files

shellixyz, 2018-03-12:
For the external baro to work you need to flash the ANYFCF7_EXTERNAL_BARO target. What do you mean by "the mag declination is not working" ?

Dhananjaya96, 2018-03-19:
thank you for reply

Dhananjaya96, 2018-03-19:
sir I already flash that 1.9.0 anyfc external baro target . still mag declination is not working .I am using hmc5883 compass and there is actual declination of -10 degree . then I open the configuration tab and set mag declination to 10 degree . and then save and reboot .still there is no effect . then I try it using cli tab .nothing happen .

Dhananjaya96, 2018-03-19:
one more question . previously I am flying my hexacopter  with I nav 1.7.3 external baro . and it feels so solid and rigid in air and return to home is extremely precise but lack of osd and sd card blackbox. but in 1.9.0 anyfc external target sd card blackbox is working for first arming then shows fatal error reboot to retry its manageable and osd looks awesome . but flying experience is not good at all while in return to home machine vibrate a lot .same with poison hold . why this happen

closed 2018-06-01 by stale[bot]