iNavFlight/inav #2910: Servo configuration broken

Arvycka, 2018-03-12:
### Board and Version
Omnibus F3 Pro
1.9.0
INAV/OMNIBUS 1.9.0 Mar  5 2018 / 22:30:20 (dc4ef59)
Configurator 1.9.2

### Behavior
Servo configuration is broken:
a. if anything is edited in "Servos" tab either in "live" mode or edited and saved for any of the servos they "lock-up" - stop responding to TX inputs or any further configurator edits. A board reboot is required.
b. Trimming does not work - if mid position is changed servo does not move to that new centre position - only its travel becomes limited in that direction.

Tested on 2 boards.

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/WyWnH74x

vavisimo, 2018-03-12:
I have this issue on omnibus f4 pro v2, i tought that i have somethinf wrongly configured, after every change in servo tab (eg. mid position on one servo) i hit save and the second change cant be made until reseat of usb cable.

Arvycka, 2018-03-12:
OK...seem to be configurator issue.
Flashed back 1.8.0 ant tried with configurator 1.9.2 - servos still freeze after editing
Then tested with configurator 1.8.1 - servos do not lock-up, end points and direction/rates can be edited continuously.

Servo MID point setting however does nothing at all - editing it does not produce any effect on that servo - a separate issue?

digitalentity, 2018-03-18:
This is the issue in between Configurator and firmware (see https://github.com/iNavFlight/inav-configurator/issues/389 for Configurator-related issue).

When "Save" button is clicked in Configurator smix rules are sent to firmware and reloaded regardless of mixer type - thus evvectively disabling servos on unmodified AIRPLANE mixer.

digitalentity, 2018-03-22:
Fixed by #2938

Arvycka, 2018-03-22:
Tested, works perfectly, thank you!

closed 2018-03-22 by Arvycka