iNavFlight/inav #2912: Matek f405 OSD INAV 1.9 regular plane servo mixer

mrkidvn, 2018-03-13:
My Sky Surfer with Matek F405 OSD worked fine with INAV 1.8.0
I upgraded firmware to INAV 1.9 and my ruder not working anymore
Servo output
S1: Motor
S2: Ruder
S3, S4: Aileron
S5: Elevator
S6: Servo gimbal
What's the problem?
I also have plane wave up and down when doing RTH.

shellixyz, 2018-03-13:
S5 (PA15) has been disabled in 1.9.0 for use with a servo (Fixed wing). No sure why.

For the waving issue please read this: #2506

shellixyz, 2018-03-13:
The removal of one servo output was accidental it will be fixed in the next release. In the meantime you can use this custom build in which the servo output for fixed-wing has been enabled again on S5:
[inav_1.9.0_MATEKF405OSD.hex.zip](https://github.com/iNavFlight/inav/files/1807102/inav_1.9.0_MATEKF405OSD.hex.zip)

digitalentity, 2018-03-14:
Fixed

PR #2916 (2018-03-14): MATEKF405: Enable FW servo on S5
Fixes #2912
closed 2018-03-14 by digitalentity