iNavFlight/inav #2913: Big difference in GPS precision when using INAV compared to u-center

swemonstro, 2018-03-13:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
Omnibus F4 Pro V3 (https://www.unmannedtechshop.co.uk/omnibus-f4-pro-flight-controller-v3/)
# INAV/OMNIBUSF4PRO 1.9.0 Mar  5 2018 / 00:15:43 (dc4ef594f)

GPS:
https://www.banggood.com/NEO-M8N-GPS-Module-for-PX4-PIX-APM2_52_6-Built-inCompass-p-974360.html?cur_warehouse=CN

_

### Behavior
I was during my first flights with GPS hold experiencing some unexpected lateral movement when using Nav PosHold. and after looking deeper at the logs from (another groundtest) I discovered that there was some drops in GPS covering where the GPS lost all of the signals (I did this test with Tethered Logging) and I could see three drops of all satellites and three jumps in position. not very strange. Also the signal is very noisy and untrustful as the picture will display. [Position thru INAV](https://www.dropbox.com/s/9f0r3obv29clp5m/GPS_position_thru_INAV.png?dl=0)

I did more testing and now I connected the GPS thru an arduino to my computer and running u-center to display the data, and here to my surprise the signal was much much better, less noise and no glitches [Position thru u-center.](https://www.dropbox.com/s/zb43quzph16f433/GPS_position_thru_ucenter.png?dl=0)

Another finding (when doing the test from my balcony, due to bad weather, where only the east part of the sky is visible) I found out that the approximate position is moved around 40-50 meters in a east-east-norht direction. [Wrong Position](https://www.dropbox.com/s/9tlk4oobt9vin21/Wrong_Position.png?dl=0)

All tests was with the quad stationary on the edge of the balcony. 

Does anybody have any idea why the INAV understanding of the GPS data can be so much worse that the u-center.

shellixyz, 2018-03-13:
It is likely due to RF interference with other equipment on your aircraft or noise in the power supply to the GPS unit.

swemonstro, 2018-03-13:
the GPS is powerd directl from the FC connector, perhaps I should power it directly from a PDB?

shellixyz, 2018-03-13:
You can try that. Also make sure your GPS receiver is as far away as possible from everything else particularly your receiver and VTX.

shellixyz, 2018-03-13:
OK but were all your aircraft equipments powered up when you did the test with the Arduino ?

digitalentity, 2018-03-13:
Also note that most GPSes that don't have a battery default to Pedestrian model which offers higher accuracy and requires less satellites, however doesn't allow high speeds and introduces considerable delay to the data.
INAV configures Ublox GPS to "Airborne 1G" model which requires better satellite coverage.

swemonstro, 2018-03-13:
Yes the quad was powered up with motors on low throttle..   

But perhaps it is the current from the FC that is a bit dirty..  will be back in 30 min with another test!

swemonstro, 2018-03-13:
Now I have the GPS powered directly from the PDB and I can say that the initial test from the same position looks much better. It has only been a couple of minutes but it looks much better! and the absolute position of the quad is much more accurate!  I will let it run for about an hour to get a good amount of data. 

If successful I will take it to a more open position and do a final test with motors running for an hour or so...

thanks for pointing me in the right direction...

swemonstro, 2018-03-13:
Here is an screenshot of the GPS track during about 30 minutes of tracking, on a quite open place in a city...  Think it start to look good enough to trust...  
[image]

It turned out to be a very good Idea to power the GPS from an PDB instead of the FC!!

shellixyz, 2018-03-13:
Indeed deviation looks "normal" now. The HDOP value should give you a good indication if the GPS can be used for navigation. Below 1.5 should be safe.

stronnag, 2018-04-04:
This support discussion appears to have run its course, closing.

closed 2018-04-04 by stronnag