iNavFlight/inav #2918: ADC RSSI pad assigned to resourceless pin in CLRACINGF4AIR and other STM32F405 based targets

dimamedia, 2018-03-13:
### Board and Version
[CLRacing F4 Air V2](https://www.clrpowered.com/products/clracingf4-air-v2)
INAV/CLRACINGF4AIRV2 1.9.0 Mar  8 2018 / 06:50:40 (dc4ef594f)

### Behavior
I connected L9R to F4 AIR V2 board with analog RSSI to its dedicated pad and enabled RSSI_ADC. 
But on OSD the RSSI reading shows 27 or 28, even when the transmitter is off.
After some researching found out that the RSSI pad on the board is connected to pin, that has no resource assigned to it.
So technically:

RSSI pad on the board is traced to STM32F405 to pin #11 - PC3.
[image]

In CLRACINGF4AIRV2 target has default: `set rssi_adc_channel = 3`

In the source:
`#define RSSI_ADC_CHANNEL                 ADC_CHN_3`
`#define ADC_CHANNEL_3_PIN               PA0`
`#define UART4_TX_PIN            PA0`

ADC Channel 3 is assigned to PA0, that is actually UART4_TX_PIN and the miserable PC3 is FREE?! 

So RSSI pad is just a dump pad, with no resource assigned.

### Solution
Move ADC_CHANNEL_3_PIN to PC3.
`#define ADC_CHANNEL_3_PIN               PC3`

As there is no resource remapping possibility in iNav, this should be changed in source.
This problem probably exists also in other STM32FR05 based targets.

bnn1044, 2018-03-14:
the FW is updated and post on www.clrpowered.com under CLRACINGF4AIRV2 products page download the lastest FW.

dimamedia, 2018-03-14:
Thanks @bnn1044! I've tested it and it worked :)

So, this can be implemented also in other targets having the same symptoms with stm32f405 chip.

I think, this can be fowarded for pull request.

shellixyz, 2018-03-18:
@dimamedia Can you confirm your modification is working ?

dimamedia, 2018-03-26:
@shellixyz yes, RSSI signal is passing through to OSD

shellixyz, 2018-03-26:
The RSSI ADC channel pin has been changed to PC3 (66bddee0062a4c145caa49c1f1abed58866769a8). Fixed.

fiam, 2018-05-25:
Fixed in #2933, closing.

closed 2018-05-25 by fiam