iNavFlight/inav #2922: Current always zero in blackbox log

StefanRvO, 2018-03-14:
I'm using blackbox logging on the MATEK F722 ( 1.9.0), but is having an issue with the current in the log (amperageLatest) always showing zero. I have current monitoring set up and the consumption shows correctly when viewed through the configurator.
I have also tried logging using betaflight, and here, the value is logged as it should.

iNav blackbox log:
[LOG00012.TXT](https://github.com/iNavFlight/inav/files/1811283/LOG00012.TXT)
Betaflight blackbox log:
[LOG00011_betaflight.txt](https://github.com/iNavFlight/inav/files/1811294/LOG00011_betaflight.txt)

diff:

[inav-13-09-2018_diff.txt](https://github.com/iNavFlight/inav/files/1811308/inav-13-09-2018_diff.txt)

shellixyz, 2018-03-18:
Please post the full configuration dump. My guess is that you didn't enable the current sensor in the configuration tab. You can see the measured current in the setup tab in the right column "Current draw" and also the the configuration tab in the "current sensor" section.

StefanRvO, 2018-03-18:
Here is a dump:
[inav-dump-18-03.txt](https://github.com/iNavFlight/inav/files/1822844/inav-dump-18-03.txt)

shellixyz, 2018-03-18:
Your current sensor have not been calibrated (you are using the default scale). Can you confirm you are seeing current values > 0 in the configurator or OSD ?

StefanRvO, 2018-03-18:
I don't have any FPV stuff, so i don't know about OSD, but the configurator shows a non-zero current.
[image]
I know that it is not calibrated, but would think that some value should be logged regardless.

shellixyz, 2018-03-18:
Indeed it should. It will be fixed in the next version. In the meantime you can use this fixed build:
[inav_1.9.0_MATEKF722.hex.zip](https://github.com/iNavFlight/inav/files/1822880/inav_1.9.0_MATEKF722.hex.zip)

mistyk, 2018-04-05:
I've the same problem with Matek F405 could you send us a Fix ? Thanx a lot

shellixyz, 2018-04-05:
@mistyk Sure: [inav_1.9.0_MATEKF405.hex.zip](https://github.com/iNavFlight/inav/files/1880271/inav_1.9.0_MATEKF405.hex.zip)

shellixyz, 2018-04-06:
The patch is available here: #2940 
Fix binaries for all targets: [iNav_1.9.0_fixed_BB_current_all_targets.zip](https://github.com/iNavFlight/inav/files/1884058/iNav_1.9.0_fixed_BB_current_all_targets.zip)

dereb, 2018-04-06:
Thank you very much!

digitalentity, 2018-04-10:
Fix merged. Closing

PR #2940 (2018-03-19): Fix blackbox current logging
Logged current is always 0 in 1.9.0 (my fault). Fixes #2922
closed 2018-04-10 by digitalentity