iNavFlight/inav #2923: No Resource Remapping?

CradyLab, 2018-03-15:
I see where SOFTSERIAL can be enabled as a feature but I'm unable to free-up an unused motor pin to use with IRC Tramp.  Is resource remapping undocumented or just unavailable?

iNAV 1.9 and Omnibus F4 v3

shellixyz, 2018-03-15:
It is not available

FinalX1992, 2018-03-15:
i have the same thing :( its not yet implented in inav, maybe it will come later im also an omnibus user, betaflight has this function but inav doesnt have it yet, im using uart 1 for gps uart 3 for compass and uart 6 for sbus i need 1 more for frsky telemetery

CradyLab, 2018-03-15:
OK.  Thanks, shellixyz.  

FinalX1992:  If you have an Omnibus F4 V3 you might want to try the [nav_1.9.0_OMNIBUSF4V3_I2C_REMAP.he](https://www.sendspace.com/file/ej7gvg)x firmware found near the end of this thread: [https://github.com/iNavFlight/inav/issues/2658](https://github.com/iNavFlight/inav/issues/2658)

I appears to give you a fully functional SOFTSERIAL port. I haven't experimented with it much but it did appear to disable the LED strip feature.

For the now the best option may be to switch to the Matek F722-STD board.  It has 5 UARTS available and it is supported in iNAV 1.9

FinalX1992, 2018-03-15:
i dont have the F4V3 i have the F4V5 so thats the issue that the FW wont work with it did tho XD

artempolikarpov, 2018-03-25:
Resource remapping is extremely useful feature. Please implement it

shellixyz, 2018-03-25:
@artempolikarpov Please see this discussion about it: https://github.com/iNavFlight/inav/issues/1145

GS-AutoTech, 2018-04-01:
I have a F4V5 original omnibus board I'm installing into a fixed wing plane. Using iNAV, Wraith32 65A ESC, Frsky X8R, DJI NAZA GPS.
Not paying attention, I was going to use UART 6 for the GPS, RX 1 for ESC telemetry & TX 3 for Smart Port. SBUS needs UART 6 so I can move GPS to UART 3. Can I use UART 1 TX 1 for Smart Port & RX 1 for ESC telemetry?

GS-AutoTech, 2018-04-01:
If I can use UART 1 for both Smart Port RX & ESC telemetry on TX,  how do I remap in CLI if remapping is needed? Set UART 1 to soft serial & bidirectional?

GS-AutoTech, 2018-04-01:
The telemetry RX and TX  would not need high priority as they are both used just to monitor system status.
I need all 6 PWM channels for servo output on the fixed wing. 1 & 2 allocated for ESC,  3,4,5,6 for elevator, ailerons,  rudder & flaps

DzikuVx, 2018-04-04:
Duplicate of #1145 Closing

closed 2018-04-04 by DzikuVx