iNavFlight/inav #2926: F35 Lightning Servo Problem with V1.9

Painless360, 2018-03-15:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Furious FPV F35 Lightening
_INAV/FF_F35_LIGHTNING 1.9.0 Mar  5 2018 / 01:09:23 (dc4ef594f)

### Behavior
* _The servos stop moving after a short time or if you try and change something in the Servo Tab. Servos can be enabled again for a short time by rebooting the FC by either removing all power or rebooting using CLI tab.
* _Power FC, use Manual mode, ensure servos are moving. Simply wait or enter Servo tab and change a few settings with LIVE turned on.
* _Servos should initially move and respond to the radio or to the new centre values as they change in the Servo Tab
* _Servos stop moving but the motor output appears unaffected. Reboot clears.

_Is this a new feature request?_Nope

#### Blackbox logs/Config (.ZIP) where problem occurs
_https://pastebin.com/cTmsdiMn

vavisimo, 2018-03-15:
same as [https://github.com/iNavFlight/inav/issues/2910](2910)

binleo, 2018-03-23:
Look like it is from app or fw, because I have the same issue when using Servo tab but when I flying, almost ok

digitalentity, 2018-03-23:
Fixes to Configurator (https://github.com/iNavFlight/inav-configurator/pull/393) and firmware (#2938) are in place. Closing.

closed 2018-03-23 by digitalentity

Painless360, 2018-04-15:
If the fix is in place - please release it..

digitalentity, 2018-04-16:
@Painless360 just did. Get Configurator 1.9.3 and enjoy!