iNavFlight/inav #2927: suggest add Compass interference Warning to OSD message

sincoder, 2018-03-15:
when compass was interference , the quad just lose control in GPS HOLD mode (very dangerous), suggest show a warning message on the OSD to tell pilot that there are interference and the compass can not work , and It is best to exit GPS HOLD mode at the same time .

DzikuVx, 2018-04-25:
The problem here is that we do not know by compass readouts if there are interferences, so I'm not sure we can determine if such a thing is happening

sincoder, 2018-04-25:
when in air meet compass interference for a while ,when auto exit HOLD mode, These are the functions that modern flight control should have

Acanevaex, 2018-05-15:
Someone would most likely have to write an algorithm to check compass heading change versus gyro yaw rate.

digitalentity, 2018-05-15:
I have an  idea about compass interference detection/warning. Will make a PoC implementation soon

closed 2023-10-27 by DzikuVx