iNavFlight/inav #2928: 5a2299a9 breaks ACC on MATEK F405

stronnag, 2018-03-15:
Since a818e2ebf933d13ffa02b42a3dcdbec44b5e0b21, the acc fails to be shown as working on MATEK F405. Running:
````
git bisect start
git bisect bad
git bisect good a818e2ebf933d13ffa02b42a3dcdbec44b5e0b21
````
results (of course) in the maximum iterations until:
````
$ git bisect bad
5a2299a99c69d31997693999daf998267a5cd638 is the first bad commit
commit 5a2299a99c69d31997693999daf998267a5cd638
Author: Konstantin Sharlaimov (DigitalEntity) <konstantin.sharlaimov@gmail.com>
Date:   Wed Feb 28 16:08:46 2018 +1000

    Add bulk MPU gyro reads; Add temperature reads

:040000 040000 5909a03c7eff40b8644ad7cf04f930e3e3089f5c 3ac05726ae3f27eba43f3bf84f2a930d5e4852ef M	src
````
Note that the branch `de_quaternion_math` has always worked, as it branched prior to the above commit.

Failed CLI status:
````
# status
System Uptime: 14 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 0.0V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=UNAVAILABLE, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1024
I2C Errors: 0, config size: 3496, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1005, PID rate: 995, RX rate: 49, System rate: 9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI
````
Good CLI status:
````
# version
# INAV/MATEKF405 1.9.1 Mar 15 2018 / 18:36:16 (a818e2eb)

# status
System Uptime: 412 seconds
Current Time: 0000-01-01T00:00:00.000+00:00
Voltage: 0.0V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1024
I2C Errors: 0, config size: 3496, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1001, PID rate: 999, RX rate: 49, System rate: 9
Arming disabled flags: RX CLI
````

stronnag, 2018-03-15:
Note this only affects the MATEK board in my fleet: SPRF3, OMNIBUS F3 and QUARKF4 all bench test correctly with 954210b8

digitalentity, 2018-03-16:
Found a bug, fix pending

digitalentity, 2018-03-16:
Fixed

PR #2930 (2018-03-16): Fix bug in MPU6500 accelerometer detection
Fixes #2928
closed 2018-03-16 by digitalentity