iNavFlight/inav #2931: implementation of ROI - POI

chris669, 2018-03-16:
Objective of iNav was "improve the current "official" Cleanflight with advanced navigation functions". I think I could be also a perfect solution for photography purposes, so why not adding ROI functionalities?

Schalonsus, 2018-03-27:
This is requested long time ago, so +1 for adding this option

chris669, 2018-04-04:
@stronnag, don't worry, will go back to BetaFlight... iNav seems to be EOL.

fiam, 2018-04-04:
@chris669 Your comment makes absolute no sense. The scope of this issue is already covered by another older one and there's no good reason to keep multiple open issues for the same feature. If you want to contribute something constructive, do so in #154.

Schalonsus, 2018-04-04:
nice, one iNav android developer less....
iNav seems really EOL atm

fiam, 2018-04-04:
@Schalonsus ?

neagron, 2018-05-08:
could we please have ROI/POI and circle this would make iNav waypoints so much better aqnd certainly more fun for me,im sorry I cant do any programing