iNavFlight/inav #2932: motors stop in air osd : aircraft not level

teamnanopirate, 2018-03-17:
Hello,

I have just crash my drone, motors stop in air, i don't know why.
All was good as you can see on vidéo link below.
Just one thing can help to found what happened, OSD message : aircraft not level.
I found my drone with nothing broken, i test it and all is ok.
unfortunaly i don't have register black box.
Anyone know what is happened ? 

https://www.youtube.com/watch?v=magwyhLrpxM&feature=youtu.be

Thanks

fiam, 2018-03-17:
Which RX are you using?

teamnanopirate, 2018-03-17:
hello,
Frsky r9 slim with last firmware 
2018-02-07 | 180202
Sbus

fiam, 2018-03-17:
@teamnanopirate The problem was not failsafe, it looks like it disarmed mid air. Are you sure you were running the final 1.9 an not a release candidate for 1.9? Please, paste your config dump so we can have all details.

fiam, 2018-03-17:
@teamnanopirate Take a look at https://github.com/iNavFlight/inav/issues/2827 and see if you can still reproduce the problem

teamnanopirate, 2018-03-17:
thanks, I'm going to look

fiam, 2018-03-17:
Ah, sorry, didn't realise you said 1.8. As far as we know, there was a problem with the R9 slim but only on a release candidate for 1.9. Still, the most likely culprit seems to be that for some reason (bug in the RX or the FC) the arming channel was lowered for at short amount of time but > 150ms. That caused the aircraft to disarm and then you couldn't arm because the small_angle protection kicked in.

sincoder, 2018-03-17:
crash as this before ,I am use R9 slim too, what I am fix it is  use "stick Arm" ,do not use "switch  arm"

teamnanopirate, 2018-03-17:
Thank you all for your help, 
I will flash the 1.9 and in the CLI put the variable small_angle = 180
After I'm going to change the setting of the R9M in Taranis 8ch by 16ch. it should work.
Otherwise I'll arm with the sticks.
I'll tell you if it solved the problem.

digitalentity, 2018-03-17:
I'd say it's a normal disarm. Possibly a momentary glitch in ARM channel - maybe a receiver issue or misconfigured failsafe.

OlivierC-FR, 2018-03-21:
I've done a lot of tests with the R9 and R9M and R9Slim units, witht the current firmwares I consider these to be highly unreliable, mostly all the modes with telemetry. I've also experienced unexpected failsafes and weird disarms without the failsafe signal being sent proper to iNav over SBUS. In short, I don't this iNav is to blame when a R9 is involved.

teamnanopirate, 2018-03-22:
maybe new Firmware fix it ? 
https://www.frsky-rc.com/r9m/

OlivierC-FR, 2018-03-22:
Maybe... but for now there's not changelog, so good luck to anybody who wants to tests this. I may do it later this week, but to be honest I'm a bit tired to see my quads fall out of the sky because of the R9, so I'm not in a hurry.

digitalentity, 2018-03-22:
I guess we can close this. It's definitely related to R9.

closed 2018-03-22 by digitalentity