iNavFlight/inav #2934: Battery warnings not displayed on OSD in 1.9

Hwurzburg, 2018-03-17:
Omnibus F4 Pro
INAV/OMNIBUSF4PRO 1.9.0 Mar  5 2018 / 00:15:43 (dc4ef594f)
I just setup and flew a new plane with iNav 1.9.....
I had set the following battery and warning values:

set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 320
vbat_warning_cell_voltage = 350

battery_capacity_unit = MAH
set battery_capacity = 1400
set battery_capacity_warning = 350
set battery_capacity_critical = 140

as you see in the video, no warning was given for either of the two capacity warnings nor for low battery voltage....
I dont think there is anything wrong in the setup so this may be a serious bug....

https://youtu.be/DAdRSex8bdw
 

iNav battery warning failures (0 min 41 sec)
--

shellixyz, 2018-03-18:
I could not reproduce the issue with v1.9.0. When the voltage goes bellow `vbat_warning_cell_voltage`   the voltage display blinks and when the remaining capacity goes below the set warning the capacity related OSD item blink.

I know this is not the issue as in your video your capacity goes below 15% but also note about the capacity warning that with your settings the capacity related OSD items will blink when below 17% remaining capacity not 25% because the capacity percentage is calculated from the remaining capacity and the remaining capacity is calculated from `battery_capacity` and `battery_capacity_critical`. So the real usable capacity with your configuration is 1400 - 140 = 1260mAh and the remaining battery percentage will display 0% when 140mAh is left in your battery. When 350mAh is left in your battery the remaining battery percentage will display (350-140)*100/(1400-140) = 17%.

Hwurzburg, 2018-03-18:
the only thing I can think of is that it is trying to use hardware blinking, and for some reason,my FC (even though its a AB7456 chip on genuine Airbot FC) is not able to do it....I will try again with display_software_blink=ON and see if it works...
I think it would be helpful if the CLI doc would explain this parameter...its not included there yet...

also, the capacity stuff is very non-intuitive and the wiki unclear....
battery_capacity_critical is really a reserve...
and the config tool implies that the warning should occur at a % value instead of an absolute mah value...changing this would help avoid the percentage confusion...as would putting an example calculation in the wiki...ie
_battery capacity = 1400mah
battery_capacity_warning= 25% of capacity or 350mah
battery_capacity_critical= 10% of capacity or 140mah
warning will flash when 350mah is remaining in battery and battery percentage used in OSD displayed is (350-140)*100/(1400-140) = 17%
when  140mah is remaining, it will display 0%_

shellixyz, 2018-03-18:
I will see what I can do to make this more clear. Also perhaps we should make the warning percentage in the GUI to be relative to the available capacity not the real capacity. Then the OSD will show this percentage when starting blinking. What do you think ?

Hwurzburg, 2018-03-18:
yes that would be better I think...I think most would use the critical capacity as a reserve(like a car gas tank) and then think of it terms of using 100% until x% warning of the non-reserve tank capacity (in cars its like at 10% of non-reserve)

shellixyz, 2018-03-18:
I don't think my GUI mod idea is good. Values in the GUI need to stay relative to specified battery capacity. What is displayed in the other hand will always be relative to `battery_capacity_critical`. Users need to see that as the capacity they don't want to go under, considering the battery empty below this capacity. The goal was to make it work just like `vbat_min_cell_voltage` but for battery capacity. But you are right something could be added to the doc to explain that the warning percentage will not be equal to the percentage set in the GUI. And we possibly could add an explanation in a GUI's battery_capacity_warning field tooltip.

digitalentity, 2018-03-19:
While #2944 adds some clarity to docs, I think we should ignore capacity checks if current sensor is not present (even if capacity thresholds are set).

shellixyz, 2018-03-19:
@digitalentity It is already:
```
batteryUseCapacityThresholds = feature(FEATURE_CURRENT_METER) && batteryFullWhenPluggedIn && (batteryConfig()->capacity.value > 0) &&
                                           (batteryConfig()->capacity.warning > 0) && (batteryConfig()->capacity.critical > 0);
```

digitalentity, 2018-03-19:
Oops, missed that 😄

Hwurzburg, 2018-03-19:
verified...warnings flash if I set software blink on....I dont understand why the hardware blink doesnt work...its a genuine F4 Omnibus with AB7456 from Airbot(via RTFQ, the disti)....
in any case, I think it might be good to put a tool tip in the batt config section of the GUI that the warning blinking on some boards may not work unless display_force_sw_blink = ON is set..

closed 2018-03-19 by Hwurzburg

shellixyz, 2018-03-19:
Good news. Odd indeed. They might have got a batch of fake OSD chips I don't know. At least you can make it work with the software blink it has been added exactly for this case.