iNavFlight/inav #2939: smix no function after reboot 1.9.1

Tribal78, 2018-03-18:
Hi,

I´m using a Matek F722 with Inav 1.9.1. I set up some smix rules in the servo tab. After saving I can see the proper function in the Motor tab.
But after a reboot there is no function, although I can see the mixer in servo tab and the cli. Pressing again save in ther servo tab enbales the mixer until the next reboot.

Any ideas?

BR
Sven

shellixyz, 2018-03-18:
Did you check the custom mixer is selected in the configuration tab after reboot ? We already know there is a bug with the custom mixer configuration in the servos tab. It is possible that when you save the mixing rules it changes temporarily to the custom mixer but the mixer selection is not saved. I will take a look later in the day.

Tribal78, 2018-03-18:
No.. didn´t checked this. I´ve just tried to find this option but I cannot find someting similar in the configuration tab? Configurator is 1.9.2.

What is the CLI command to enable the mixer?

Thanks!!

shellixyz, 2018-03-18:
The CLI command for changing the mixer is `mixer`. Used without argument it will show the current mixer. For selecting the custom airplane mixer: `mixer CUSTOMAIRPLANE`.

[image]

Tribal78, 2018-03-18:
Okay. But my purpose is to add a new mixer in order to move both ailerons up and down.
Therefore I´ve added some smixer. But as mentioned they work only directly after saving but not after a reboot?
Is there a Special comand to enable These smixer? By typing smix in the CLI they are correctly shown but for some reason they are not working.

Entering CLI Mode, type 'exit' to return, or 'help'

mixer
Mixer: AIRPLANE

smix
smix 0 3 9 97 0
smix 1 4 9 -99 0

shellixyz, 2018-03-18:
Oh OK now I understand what you are trying to do. For the moment you can't just add mixer rules. You need to select the `CUSTOM AIRPLANE` mixer and write all the rules.

Tribal78, 2018-03-18:
Thanks for this Information! Properly saved me a lot of time :-)
So it is not possible to use smix while the Mixer AIRPLANE is running?

Is somewhere documented what the AIRPLANE-mixer is doing? So I can use this as a base for my CUSTOM AIRPLANE mixer.

shellixyz, 2018-03-18:
No it is not possible. Mixers are not editable. It is planned to move mixers to presets but it hasn't been done yet. When it is done you will be able to see the mixing rules for the preset and alter them.

It is documented in the source code ;)

The default mixer for airplane is this:
```
smix 0 3 0 100 0
smix 1 4 0 100 0
smix 2 3 14 100 0
smix 3 4 14 -100 0
smix 4 5 2 100 0
smix 5 2 1 100 0
```

Tribal78, 2018-03-18:
Thanks again! I´ll try this.

closed 2018-04-05 by DzikuVx