iNavFlight/inav #2943: No GPS RTH/PH mode and no connect with Tower/Mission Planner

WernerSch, 2018-03-19:
Hello

i hvae a Starracer StarF3S board 
https://www.banggood.com/Racerstar-StarF3S-30A-BL_S-4-In-1-ESC-5V-BEC-w-F3-Flight-Controller-AIO-OSD-BEC-Current-Sensor-p-1157101.html?rmmds=search&cur_warehouse=CN

First i installed 1.8.0 and now 1.9.0. I have take the Omnibus Board.
I have a UBlox Neo GPS Modul installed and get a GPS fix in the inav Chrome App. But in the Mode Tab i dont have the Nav modes RTH or PH. 
For Programming Waypoints i get no connect with Tower(ex Droid Planner) or Mission Planner.

best regards
Werner

shellixyz, 2018-03-19:
Please post a full config dump. I don't think Tower supports MSP so you can't use it with iNav. About Mission Planner if you are talking about the one Published by Eziosoft it should work.

WernerSch, 2018-03-19:
ok, when i am at home i send a full dump

meanwhile I found  a site where note says for RTH and PH i need a compass too.
Is that true? My Board has no I2C Bus for a Compass
can i solder 2 wire to the µC? Which Pins must i use?

closed 2018-03-19 by WernerSch

shellixyz, 2018-03-19:
For a multirotor you need a compass for a plane you do not. It is documented in the wiki. If the MCU is an STM32F303 in LQFP48 package you can probably solder to PA9 (SCL) and PA10 (SDA) pins if they are not used already but you will need a custom build of the firmware with the I2C and compass drivers enabled.

WernerSch, 2018-03-20:
ok thats the Problem
Yes i have a STM32F303. I have the Omnibus Version on my Board. Omnibus normaly has an I2C Bus. The CLI cmmand resource said that the I2C Bus is on PB6 and PB7. Thats the same i reed in the Forums

shellixyz, 2018-03-20:
PB6/PB7 is I2C1, PA9/PA10 is I2C2. PB6/PB7 are the right pins (the ones normally used) for the external I2C on Omnibus so if you use these ones you should not need a custom build.