iNavFlight/inav #2957: Add roll/pitch angles to OSD

digitalentity, 2018-03-22:
Having numeric angles of roll and pitch angle in OSD will make it easier to figure out proper board alignment angles for airplanes. @DzikuVx @fiam what do you think?

fiam, 2018-03-22:
I think it’s good, specially once we have multiple layouts. Users can have one for normal flight and another one configured for running (PIDs, these angles, etc)

giacomo892, 2018-03-22:
Would be nice to add PID outputs too. Useful for example in a wing to check live how much a surface is deflected.

shellixyz, 2018-03-22:
@giacomo892 For a wing to check the surface deflection the servo outputs would be more useful (requested in #2674). But PID outputs would probably be useful for MRs.

shellixyz, 2018-06-16:
This has been done the issue can be closed