iNavFlight/inav #2959: Flaperon on INAV and Omnibus F3

Christianmix, 2018-03-23:
Hello everyone, I have a question, I searched here but did not find anything related, when I activate the flaperon of my sky surfer, the ailerons go up, how do I get them down instead of going up? Studying the theory of flaps I found that they are much more effective as brakes when they are down, how do I change that to make it go down? All the rest of the plane is working perfectly, stabilizing, RTH, Fail Safe, even the flaperon button, just want to change the direction in which it comes into action and put down the ailerons.

thank you all

Christianmix, 2018-03-23:
my dump

# mixer
mixer AIRPLANE

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 -100 -1 
servo 3 1000 2000 1500 -100 -1 
servo 4 1000 2000 1500 -100 -1 
servo 5 1000 2000 1500 -100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature RSSI_ADC
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
[... 495 lines cut ...]
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 60
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set rc_expo = 30
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 15
set yaw_rate = 13
set manual_rc_expo = 40
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

#

shellixyz, 2018-03-23:
Hello. You need to change it in the mixer. It also does the same on my planes I think it should be fixed. Also I don't know why the offset setting have been changed from negative min to 0. It would allow to change the direction without changing the mix.

By default the flaps are mixed like that:
```
smix 2 3 14 100 0
smix 3 4 14 -100 0
```

You need to change that to:
```
smix 2 3 14 -100 0
smix 3 4 14 100 0
```

Christianmix, 2018-03-23:
How do I change this?

shellixyz, 2018-03-23:
```
mixer CUSTOMAIRPLANE
mmix reset
mmix 0 1 0 0 0
mmix 1 1 0 0 0
smix reset
smix 0 3 0 100 0
smix 1 4 0 100 0
smix 2 3 14 -100 0
smix 3 4 14 100 0
smix 4 5 2 100 0
smix 5 2 1 100 0
```

Christianmix, 2018-03-23:
thank you very much, solve it :)))))))

Christianmix, 2018-03-23:
Let's see if I understand, this link that you have posted now, will it be the next amendment that will solve the flaperon problem?

Christianmix, 2018-03-23:
Not wanting to abuse your goodwill, but already abusing, I bought a mini talon and I can not make the vtail mix work, you would like to send me the code as you sent me from flaperon, I wanted to use it with flaperon too, I I took the omnibus f4 v3 and put an apm, because I could not understand myself with the mix of the servos.

shellixyz, 2018-03-23:
Did you try the mix for the Mini Talon provided in the wiki ? https://github.com/iNavFlight/inav/wiki/Custom-mixes-for-exotic-setups#v-tail-fixed-wing

digitalentity, 2018-03-24:
Fix for flaperon direction merged. Closing

PR #2961 (2018-03-24): Fix flaperon mode mix direction and widen flaperon_throw_offset min/max
Fixes #2959
closed 2018-03-24 by digitalentity