iNavFlight/inav #2963: Starnge behavior when activating MSP on Matek F722

Antiheld86, 2018-03-23:
### Board and Version
Matek F722-STD
# version
# INAV/MATEKF722 1.9.0 Mar 18 2018 / 17:09:41 (dc4ef594)

Custom Hex from #2922 
https://github.com/iNavFlight/inav/files/1822880/inav_1.9.0_MATEKF722.hex.zip

### Behavior
When activating MSP on UART4 motors react strange. Motor 1 and 3 work normal. Motors 2 and 4 don't spin up at first, but when motors 1 and 3 spin up to max and then down again motors 2 and 4 give full throttle and don't stop till battery disconnect. Even disarm won't stop them.

shellixyz, 2018-03-23:
Might be fixed in the development branch. Could you please try with this hex. Just remember if you try to fly with this hex that it is from the development branch and not as well tested as releases you might have other issues with it.

[inav_1.9.1_MATEKF722.hex.zip](https://github.com/iNavFlight/inav/files/1843342/inav_1.9.1_MATEKF722.hex.zip)

Antiheld86, 2018-03-23:
Its working correctly. Thank you!

shellixyz, 2018-03-23:
Good to know thanks for testing

closed 2018-04-25 by DzikuVx