iNavFlight/inav #2968: Servo Output Freezing

Lusby, 2018-03-26:
Board: Omnibus F4 Pro V3
 
Version of INAV: OMNIBUSF4PRO 1.9.0 Mar  5 2018 

Configurator 1.9.2

Servo output freezes when servo values are altered on servo tab in configurator. This occurs when the values are changed and saved,  output also freezes when live output mode is selected. The servos do not respond until the board is rebooted. 

Previously in version 1.8 on the same board, the servo control was re-established after a short delay when values were changed and saved, and live servo output worked also.

DIFF:
https://pastebin.com/3r3DNMza

shellixyz, 2018-03-26:
Duplicate of #2910. A fix for this issue has already been merged. As a workaround you can either use a custom mixer or use the version 1.8.1 of the configurator to change the servo parameters.

digitalentity, 2018-03-27:
Closing as a duplicate

closed 2018-03-27 by digitalentity