iNavFlight/inav #2975: Current Sensor shows maximum 29.0 A with MWOSD 1.7 and INAV 1.9

stagesixcup, 2018-03-27:
### Board and Version: F3 Racing Board with INAV 1.9 Ranger EX (Fixed Wing)

### Behavior

Hi,

i am very happy with INAV so far. Its getting better and better. 

i noticed since the new Scale for Voltage&Current Monitoring that i have some strange issue with MWOSD 1.7. Everything else runs fine, (Flight Modes, GPS, Compass, etc...).

I have the HGLRC AMASS XT60 120A Current Sensor. (See Link below) The OSD shows only a maximum of 29.0 A regardless of throttle position, also in Flight summary. (Until that value all mesaurements are fine and correct), measured with Wattmeter it shows correct 50-60A on Full Throttle. I tried to figure out what the problem is, i think it has something to do with the new Scaling? 

I was searching also for the **"multiwii_current_meter_output | OFF" Variable** in the CLI, but the option is not existing anymore.

I configured a Scale from 1145 for Current Meter (ADC) connected to Pin 7. 
I Can also provide today a CLI of my setting!

**Current Sensor:**

[https://www.banggood.com/HGLRC-AMASS-XT60-25_2V-6S-120A-Current-Sonsor-for-RC-Drone-FPV-Racing-p-1252742.html?rmmds=search](url)

thanks in advance!

regards

shellixyz, 2018-03-27:
The current sensor scale have not been changed. 1145 for the current meter scale seems very high. It means your current sensor outputs 114.5mV/A. Since the ADC can only measure up to 3.3V it means if this scale value is right for your current sensor that your FC will only be able to measure up to 3.3/0.1145 = 28.8A. This matches your issue and is perfectly normal. Your issue is either that your current sensor is misconfigured (wrong scale) or that you need to scale the output voltage of your current sensor down for the FC to be able to read its output above 28.8A.

stagesixcup, 2018-03-28:
Thank you for the fast hint, correct my Current Meter scales 1/10 (So 60A are 6.0Volts) and i can't change this anyway, i tried to solve it with a voltage divider but no Voltage can be measured anymore, maybe it is faulty. i will use the Virtual Sensor instead.

Thank you it can be closed!

closed 2018-03-28 by stagesixcup

shellixyz, 2018-03-28:
You current sensor is most likely fine but you can't just use a simple resistive voltage divider because it effectively changes the sensor amplifier factor. You could use a simple resistor and pull the output down to change the scale as you wish but it is not very practical because you will need high resistor values like 1Mohm.

Your current sensor is probably based on an INA138:
[ina138 high side current shunt amplifier.pdf](https://github.com/iNavFlight/inav/files/1856180/ina138.high.side.current.shunt.amplifier.pdf)

Alternatively you can build this circuit that would give you a max measured current of 63A (based on your 114.5mV/A scale):
[image]

stagesixcup, 2018-03-28:
thank you! i have already bought a APM Power Meter with 90A for 4 € with ADC 5V (So it should have 5V at 90A), here should also work the Voltage divider to 3.3V. If not, no problem 3,3V is 60A which is more than enough more my build.

[https://www.ebay.at/itm/90A-APM2-5-APM2-52-AttoPilot-Voltage-Current-Sensor-Module-APM-Flight-Controlle/182260697612?hash=item2a6f958a0c:g:wH8AAOSwvzRXxGoZ](url)

so, normally i would solder the circuit above but i have no time at the moment. 

But maybe this helps anyone else!