iNavFlight/inav #2979: INAV Configurator 1.9.2 modes not showing

Ebonezer, 2018-03-27:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
Matek F405-CTR
_Version of INAV used goes here_
1.9.2
_Use CLI command `version` and put its output here_
# INAV/MATEKF405 1.9.0 Mar  6 2018 / 18:21:38 ()
_If it's a custom-compiled firmware please mention this_

### Behavior
Hi, I am using INAV Configurator 1.9.2 on a Matek F405-CTR FC. The RTH and Alt Hold and Pos Hold are only listed in the Modes Page when the Magnetometer, Barometer and GPS are live. This means I have to power the quad from a battery, take it outside to get GPS lock and do any config in the garden! Is this normal?
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_
No
[dump 20180326.txt](https://github.com/iNavFlight/inav/files/1852230/dump.20180326.txt)

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

stronnag, 2018-03-29:
Indeed, it has never been necessary to have a GPS fix to configure nav modes.

shellixyz, 2018-03-29:
Odd. I usually do all the initial setup on my desktop computer where there is no GPS reception. I'm not using the MATEK F405 FC but it shouldn't make any difference. You should be able to configure the navigation modes (POSHOLD, RTH, WP, ...) as soon as your magnetometer is detected. No need for a GPS fix.

LibertyAloft, 2018-03-31:
I have the exact same issue with INAV Configurator 1.9.2 and a Matek F405-CTR FC.  I cannot see  any GPS or NAV modes in the modes tab until all the pre-arming checks have turned green.  This means I must have a flight battery powering the quadcopter, wait for all the pre-arming checks to turn green, then these modes will show up.

LibertyAloft, 2018-03-31:
Here's my version:
# version
# INAV/MATEKF405 1.9.0 Mar  5 2018 / 00:53:14 (dc4ef594f)

Ebonezer, 2018-04-01:
Hi, here are my findings but I don't claim to understand the reasons.  Since starting this thread I have tried combinations of firmware from INAV and Matek and different versions of configurator including Betaflight and all performed the same.  I then desoldered all the Matek board connections, cleaned the board and resoldered.  Everything now behaves as I would expect.  Connecting USB only and then launching configurator shoes that mag, baro and GPS are connected but not functioning and GPS Modes are not shown.  However powering Matek with LiPo then connecting USB shows mag, baro and GPS alive and functioning and all GPS Modes show in the Modes page of configurator without any satellites being discovered.   Talking with the boards supplier who has had experience of I2C data manipulation he did report that I2C is extremely sensitive to good (or poor!) connections and resoldering might have repaired a rather 'iffy' joint.  Who knows?
As a note: I read in the INAV "Flight Modes"  documentation it says that GPS configuration modes will not be available until a GPS lock is established - so that's not quite right either. Perhaps there's another issue with my board?
Thanks for your input and I hope this helps,
Cheers, EbS

stronnag, 2018-04-01:
The matek boards may not to supply sufficient power to peripherals when powered off USB to power external devices (one of mine does, one does not). Thus the GPS is not powered, and I2C is not happy as there's an unpowered mag in the bus. When powered properly, the sensors all work just fine. 

Not an iNav issue.

stronnag, 2018-04-04:
Closing, as this is not a firmware issue.

closed 2018-04-04 by stronnag

emekadaniel, 2018-08-25:
Hello,pls I really want to know why their is no GPS modes (POSHOLD, RTH, WP....) On my inav 2.0.0 configurator

padcom, 2019-06-19:
It is such a pity nobody would properly resolve that issue - and it is VERY COMMON!

So, when you flashed a new firmware and cleared all your settings your FC is basically clueless on what type of aircraft you have. Therefore it will not show _ANY_ nav modes even if the hardware is up and running properly.

I guess the developers of iNav _ass u me_ d that it is so damn obvious that you first need to set the aircraft up and then you will move on to setting up flight modes. This is absolutely not the case if you're just checking stuff out on the bench and that's when you will encounter this issue.
I have checked the docs for iNav through and through and I have not found a single line mentioning that nav modes will be available ONLY after the type of aircraft is defined (I suppose including servo mixers).

padcom, 2019-06-19:
> Closing, as this is not a firmware issue.

oh and yeah.. this is not firmware issue - it is _expected behavior_ that is nowhere documented.

GEARWELL, 2021-01-27:
You mus firs check al arming conds to prevent fail safe mode

[image]