iNavFlight/inav #2981: Disarm without reason in mid air

Antiheld86, 2018-03-29:
### Board and Version
# version
# INAV/MATEKF722 1.9.1 Mar 23 2018 / 21:15:18 (c60386f9)

### Behavior
I had my second flight with my new quarcopter. While in 3D hold which worked nice, just a bit wobbly the quad disarmed without any reason i can see in 13m hight with throttle 40%. The log says it Arm = off / Heading hold = on. I did not switch any mode while it happend.

#### Blackbox logs/Config (.ZIP) where problem occurs
[log fall out.zip](https://github.com/iNavFlight/inav/files/1860554/log.fall.out.zip)
[dump.zip](https://github.com/iNavFlight/inav/files/1860557/dump.zip)

Log and dump attached. In the dump are no NavModes configured because the Crash cutted one wire of the Magnetometer so These modes got disabled because of Hardware fault.

stronnag, 2018-03-29:
If my quad fell out of the sky with this much movement in the AH whilst doozing around in PH at < 3m/s and RSSI dropping < 20% at 14m from home, would one:
* Blame the firmware; or
* Fix the hardware

Tough choice.

https://vimeo.com/262424703

Antiheld86, 2018-03-29:
I'm pretty aware about the fact that my quad is not perfectly configured. And i dont know why the rssi dropped before. But i already tried the failsafe and the rth and it performed well enough. But the log says that something disabled arm and enabled headless. And it was not me. So i would like to know what it was. I am knew to this stuff and i am willing to think about that it was a failure of my configuration. But the last thing the Firmware should do, is to disarm and let it drop out of the sky. What i know is, that the failsaife wouldn't do that.

And it is very difficult to optimise PIDs or something to get flight behavior nicer if you are afraid that it just disarms because of - you don't know....

Antiheld86, 2018-03-29:
It is set to "hold". But i don't think the Signal was lost. I could give commands all the time.

Antiheld86, 2018-03-29:
I don't understand why that whould be. I could not find the reason in the log. Where do i find it? The strange thing about that is, that my Arm mode is on channel 9 and headless is on channel 14. Level, AltHold, PosHold and RTH are on Channel 10 with Logical Switches. But obviosly just channel 9 an 14 changed. Is it possible to see the channel values in the log?

shellixyz, 2018-03-29:
The disarm reason is recorded at the very end of the log. It is not displayed by the blackbox viewer. You need to open the file with a text editor. At the end of the log you will see "End of log (disarm reason:4)". 4 is disarm by switch.

```
typedef enum disarmReason_e {
    DISARM_NONE         = 0,
    DISARM_TIMEOUT      = 1,
    DISARM_STICKS       = 2,
    DISARM_SWITCH_3D    = 3,
    DISARM_SWITCH       = 4,
    DISARM_KILLSWITCH   = 5,
    DISARM_FAILSAFE     = 6,
    DISARM_NAVIGATION   = 7,
    DISARM_REASON_COUNT
} disarmReason_t;
```

Only the 4 first channels (thr/roll/pitch/yaw) are recorded in the log (rcData).

Antiheld86, 2018-03-29:
I had the disarm_kill_switch=off, so it should not have been able to disarm, am i right?

shellixyz, 2018-03-29:
Right, not if the throttle channel never went below `min_check`

fiam, 2018-03-30:
disarm_kill_switch is no longer available in the configurator for safety reasons, you need to set it in the CLI. In most setups, you shouldn’t need to fiddle with that.

sincoder, 2018-03-30:
use this tool to check your INAV config ,maybe helpful

Pairan, 2018-03-30:
@stronnag just a little off topic ... but what tool was that you used to visualise the dump?

stronnag, 2018-03-30:
@Pairan mwp https://github.com/stronnag/mwptools

Pairan, 2018-03-30:
thanks, mate :)

Antiheld86, 2018-03-30:
@fiam thanks for the clear up. It was explained different by @shellixyz in #2983

shellixyz, 2018-03-30:
@Antiheld86 Oops sorry for that.

Antiheld86, 2018-03-30:
Thank you for your help anyway.

Antiheld86, 2018-04-09:
Still don't know what the reason was for disarming was, but I also found out to have a bad X8R in terms of range. Got RSSI warnings at 10m range. Bought a new one and now range seems to be ok. I still would think failsafe should have behaved differnt but can't tell if there is an other problem with the R8X.

digitalentity, 2018-04-10:
Blackbox log or OSD will show disarm reason. Without knowing what code path caused the disarm we can't fix.

Antiheld86, 2018-04-10:
Is both attached in first post. The disarm reason in the log is 4. Disarmed by switch. But I did not turn any switch...

digitalentity, 2018-04-11:
Disarm reason 4 can only appear if receiver sent appropriate switch position to the FC. No other reason is possible. Maybe (for a minor hickup of receiver protocol handling) we should introduce a safety delay for changing AUX channels (in terms of applying a median filter on them or some other method that would prevent momentary changes in AUX positions).

shellixyz, 2018-04-11:
We could also log the channel used for arming to get a more precise idea of what is going on with it when an unexpected disarm happens.

Billj747, 2018-04-21:
Ever since I downloaded 1.9.0 the Horizon mode & AltNav hold do not maintain altitude, and the plane enters a slow dive.  The Altimeter is also extremely intermittent despite 10+ satellites.

I randomly lost arming (where the stats screen pops up) and had to ditch it in a lake.  After replacing all the electronics, including a new FC, I still had the same problems with random disarming (and resetting the "Home" point to where it (eventually) rearms.  If it rearms.  I ended up losing my plane 1 mile out due to this random disarming.

Omnibus F4 Pro V3, R-XSR, iNav 1.9.0

digitalentity, 2018-04-21:
@Billj747 did you notice the reason for disarming in OSD?

Billj747, 2018-04-23:
@digitalentity - It always says Disarmed by Switch.  But the switch was never flipped in any of the 4 times that it randomly disarmed.

digitalentity, 2018-04-23:
@Billj747 if it says "Disarmed by a switch" than the disarm signal originated from the code processing the receiver input. We've noticed this problem with R9Slim receivers, maybe a buggy receiver firmware is responsible for this. 

The real problem is that S.Bus doesn't have any protection against data corruption and a rogue noise pulse will disrupt comunication. Since a momentary AUX channel flip is enough to disarm the quad - this is a problem.

digitalentity, 2018-04-23:
Actually our RC median filtering is applied to AUX channels as well. Signal has to be really messed up for the corruption to get through.

stronnag, 2018-04-24:
Is this the same filter whose application logic was inverted in 1.9 and fixed in 1.9.1?

digitalentity, 2018-04-24:
@stronnag yes. Since the "disabled" setting was actually enabling the filter, the issue reported was actually with filter **enabled**. I wonder how many data corruptions was there on the wire if they skipped the filter eventually.

giacomo892, 2018-04-24:
@stronnag @digitalentity perhaps we should enable that filter back on AUX channels making to filter even more? We won't suffer than much in terms of latency from the AUX channels. Also if the machine is armed via stick combo it needs to be 2 channels to be in the disarm position for some time. Feels like a burst of bad PWM/PPM/SBUS (IBUS and FPort have cksum and they should be very safe to use) command can disarm a machine if the arming channel gets junk data.

stronnag, 2018-04-24:
@fiam also had a suggestion for additional sbus validation, which I'm not in a position to look up at the moment (maybe sbus channels 17 and 18 bring zero (or not))

brat002, 2018-04-24:
I also experienced such kind of problem on 1km height. That would be disaster, but I had armed again via switch.
matek f405-ctr and Frsky L9r receiver.

digitalentity, 2018-04-24:
@brat002 @Billj747 @Antiheld86 
Can you post output of `tasks` CLI command when your receiver is powered up and RC link is enabled?

brat002, 2018-04-24:
>@brat002 also S.Bus?

Yes. I thought I am alone with the problem, but it is not.

>Can you post output of tasks CLI command when your receiver is powered up and RC link is enabled?

I will do this at evening.

Billj747, 2018-04-24:
@digitalentity - unfortunately the first FC & Rx met its death in a lake.  The second FC & Rx was lost with the plane. I'm a bit hesitant to build another one after losing $500 in electronics and the plane from 4 random disarming events, only 2 of which were able to rearm and save the plane.

FWIW, they were both Omnibus F4 Pro V3 boards and R-XSR receivers with the R-XSRFCC171009.frk firmware and the uninverted SBUS signal taken from the "P" pad of the Rx.  I'm not sure if that helps diagnose anything.

digitalentity, 2018-04-24:
@Billj747 if it's hardware modified RX, I'll blame it. Small CPUs have very week output drivers and relatively long wire from hacked RX to the FC might introduce all sorts of signal disruptions. Inverter installed on those RXes not only inverts the signal, it also adds better capabilities to drive the long line.

Billj747, 2018-04-24:
@digitalentity  Do you really think adding 5mm of line to the "P" pad on a total length of 12cm MAX length of wire from the Rx to the FC would have an effect?

digitalentity, 2018-04-24:
@Billj747 it actually might. It's not about adding more length to the wire itself. It's about significantly reducing receiver's ability to drive the long wire.

Billj747, 2018-04-24:
12cm is long?  I don't have the plane anymore but it's probably closer to 8-10cm.  If that's the case, i'll pick a different FC where I have enough UARTS to use the inverted SBUS signal.

digitalentity, 2018-04-24:
It depends on how you drive the wire. In many cases 12cm would be considered a "long wire" indeed. 

Think of a signal over a long wire as of a 100kg weight you need to lift. You can try lifting it manually - possible, but unreliable (this is receiver CPU's weak pin). Alternatively you can use a crane which will make the task much easier (this is an inverter-driver on RX having stronger source/sink capabilities).

Billj747, 2018-04-27:
@digitalentity my earlier message was incorrect.   I am taking an **uninverted SMART PORT** signal from the "P" pad of the Rx (which is for S.Port.  The "B" pad is the uninverted SBUS).

Since I am connected to the inverted SBUS, do your comments still apply?

digitalentity, 2018-04-27:
@Billj747 with an invertor chip you probably don't have the noise issue and it's likely the receiver itself that misbehaves.

Billj747, 2018-04-27:
@digitalentity Has multiple people running the R-XSR have this problem?  What is known to be a reliable receiver?

Scripto23, 2018-04-28:
I don't think it is the R-XSR. I was running Crossfire (full rx) with ppm to an Omnibus F4 board, and had the "disarmed by switch issue". I was about 2 miles out and lost the whole plane. The common denominator in these scenarios seems to be the Omnibus F4.

Skaloverus, 2018-04-29:
I had the disarm issue today.  Running F35 board with R9m.  One thing as suggested by my wingman,  is to off the telemetry module on the Tx and I did just that.  Seems like the problem goes away.  Manage to fly 2 packs.  But I've no visibility of my Rssi after that.

Billj747, 2018-04-29:
@Scripto23 Then what is a popular FC that's known to be reliable?
@Skaloverus Can you not view RSSI through the OSD even with telemetry turned off?

It sounds like there is some programming work that needs to be done, and not just pointing blame at hardware.

Skaloverus, 2018-04-29:
@billj747 well if I off the tick box for telemetry,  even the rssi in my osd become 0. And the LCD bar on the transmitter disappear.  Just have to trust the r9m range and rth after that.

Nandox7, 2018-05-07:
Hey, just got this happening again and again almost lost the plane.
In the OSD the reason is the same Disarm by switch, but I have a LRS system and I was no more than 100mts and roll & pitch control remained working. Plus I tried to disarm/arm a few times without any change... :(
Is there a way to log channel output or any other data to try to understand what is happening, I had lots of flights done with FW 1.7.2 without a single failure and now that I moved to 1.8 and 1.9 this keeps happening.

mosvov, 2018-05-08:
https://youtu.be/76qdMOPEHZs?t=4m25s here same problem with FrSky XM+ receiver

valzaitsev, 2018-05-12:
Hello, I had the same problem 2 days ago. Almost lost my quad. INAV 1.9.1 (03a5c1922) MATEKF722 + R-XSR via sbus.

Nandox7, 2018-05-13:
Here's the video of the last time it happened, lucky it wasn't over the water and while I wasn't too high even trying to arm/disarm multiple times did not worked and it ended up crashing in a more or less controled way as I was still able to steer it.
https://www.youtube.com/watch?v=YE-RQ5wAkos
when I got close to the wing all was working fine and it armed again.

Billj747, 2018-05-13:
@Nandox7 at least you were not that far away and was easily LOS.  I lost my plane when it was 3Km out and as I lost altitude, I also lost LOS and then the plane.  I guess I won't be in a hurry to build another plane until this is figured out since it does appear to be a programming issue and not hardware related.  Or just go back to 1.8 which seems to be working fine.

Nandox7, 2018-05-13:
@Billj747 this time I wasn't, previous time I was 2Km out when I see it gliding disarmed and almost hit someone as I was trying to glide it down without motor. :(
Well I went back to 1.8 and still happened, the video above is with 1.8...
I'm going back to 1.7.3, that is the last version that I have used and with several hours flying did not seen this happening.

Nandox7, 2018-05-14:
I wouldn't say this is related specifically to S.BUS as I'm using EzUHF over PPM.

DzikuVx, 2018-06-05:
We strongly belive that this problem is solved with upcoming 2.0. Arming channel has to go to "disarm" value for a configurable period of time to trigger a disarm

tavdog, 2018-06-05:
I hope they implement this in Betaflight as well.

DzikuVx, 2018-06-05:
ATM we have a few hours in the air with this solution among testers of 2.0 and until now zero problems. Even with previously problematic FrSky R9 868MHz link

Billj747, 2018-06-29:
Any updates? I'm a bit hesitant to use this again until there's a lot of confidence that this issue was resolved (and that their was a programming issue).

digitalentity, 2018-06-30:
I believe a combination of #3110, #3118 and #3482 will fix the issue.

digitalentity, 2018-07-16:
Since https://github.com/iNavFlight/inav/pull/3482 is merged, RC receiver failsafe settings won't mess up processed channel values (`rcData`). I think this issue should now be fixed.

digitalentity, 2018-08-05:
Closing. Issue seems to be resolved

closed 2018-08-05 by digitalentity