iNavFlight/inav #2982: Navigation flight modes get deleted at hardware fault.

Antiheld86, 2018-03-29:
### Behavior
The fact that Navigation flight modes are not configurable when mag/gps/baro are not available deletes all already configured modes when one of these is disconnected. That makes it uncomfortable and probably dangerous if you disconnect for doing changes or something and power up the board but dont apply the modes again later.

_Is this a new feature request?_
yes

shellixyz, 2018-03-29:
I don't think it clears the navigation modes if you just unplug the GPS/mag only if you disable the GPS feature (`feature -GPS`).

shellixyz, 2018-03-29:
I checked the code. @Antiheld86 is partially right. When the mag is not detected it also makes the navigation flight modes unavailable.

fc/fc_msp_box.c:
```
    if ((feature(FEATURE_GPS) && sensors(SENSOR_MAG) && sensors(SENSOR_ACC)) || (STATE(FIXED_WING) && sensors(SENSOR_ACC) && feature(FEATURE_GPS))) {
        activeBoxIds[activeBoxIdCount++] = BOXNAVPOSHOLD;
        activeBoxIds[activeBoxIdCount++] = BOXNAVRTH;
        activeBoxIds[activeBoxIdCount++] = BOXNAVWP;
        activeBoxIds[activeBoxIdCount++] = BOXHOMERESET;
        activeBoxIds[activeBoxIdCount++] = BOXGCSNAV;
    }
```

`sensors(SENSOR_MAG)` is false when the mag is not detected. Would be good to make `sensors(SENSOR_MAG)` true when the mag is configured even if it is not detected. I will PR that.

closed 2018-06-11 by stale[bot]