iNavFlight/inav #2983: Add “Disarm motors regardless of throttle value“ switch

Antiheld86, 2018-03-29:
### Behavior
To avoid accidentally disarm in mid air it would be great to have a Feature wich does not allow disarm when throttle is above a value.

_Is this a new feature request?_
yes

shellixyz, 2018-03-29:
This already exists. You need to set `disarm_kill_switch` to `OFF`. It won't disarm unless throttle < `min_check`.

digitalentity, 2018-03-29:
As @shellixyz mentioned - such feature already exists. Closing.

closed 2018-03-29 by digitalentity

Antiheld86, 2018-03-30:
In the failsafe tab this Option is disabled in my configurator. The CLI says `disarm_kill_switch=on`

[image]
[image]

fiam, 2018-03-30:
That’s a different option. See my comment in #2981.