iNavFlight/inav #2988: ESC's initializing twice upon power up

jaywalk101, 2018-03-30:
### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

INAV/OMNIBUS 1.6.1 Ma

_If it's a custom-compiled firmware please mention this_

### Behavior
* ESC's will beep twice upon power up (Littlebee 20a multishot).  They initialize just fine once the board settles.  It seems like Inav/Omnibus is providing power to the ESC's briefly, and then restarting power after the board/firmware load all the way.
* Simply plug in battery to provide power to Omnibus F3 board and to the ESC's. 
* Should only beep once up on powerup
* Otherwise everything seems fine, however ESC behavior is concerning.

_Is this a new feature request?_

No

Extra Info --  I loaded latest Betaflight firmware on the same Omnibus F3 board, and this behavior does not re-produce itself.  I also tried the Inav 1.9 firmware and same issue occured. So it seems to follow iNav firmware.  

Thanks!

shellixyz, 2018-03-30:
It has already been fixed. The next release will not do that.

jaywalk101, 2018-03-30:
Good to hear!  Any idea when the next release is coming out?  Thanks!

giacomo892, 2018-03-31:
This was causing also fast servo movement upon boot on fixed wings. Glad it's fixed

shellixyz, 2018-03-31:
Unfortunately the servo twitching on startup is not fixed it is a separate issue we believe is caused by pull-ups on the outputs and will be harder to fix.