iNavFlight/inav #2992: inconsistencies in the documented coordinate frames and the presented values

humanoid2050, 2018-03-31:
Board name:
OpenPilot CC3d Revo from Hobbyking. https://hobbyking.com/en_us/openpilot-cc3d-revolution-revo-32bit-flight-controller-w-integrated-433mhz-oplink.html

INAV configurator version:
1.9.2

INAV firmware version:
INAV/REVO 1.9.0 Mar  5 2018 / 00:28:15 (dc4ef594f)

### Behavior
* Description: The developer info page describes a local coordinate frame of x-forward, y-right, z-up, which is a left handed frame. My personal experience tells me that the matrix math is much easier to keep self-consistent when you use all right handed frames and I think I'm not alone in that assertion (https://github.com/iNavFlight/inav/issues/50), but that's not my point here. The issue is that when I start manipulating the orientation of my flight controller, the real time display is not consistent with the documented left hand vehicle coordinate frame. 

* How to Reproduce: Connect the FC to the iNAV configurator, wait for the connection to establish and the Setup tab to be populated with Attitude information. Start rotating the FC about each individual axis.

* Expectations: For a left hand frame as described in the documentation, the rotation directions should be as follows:
axis    | label | direction of positive rotation
-----------------------------------------------------------
fwd     |  x     | right side UP
right   |   y     | nose down
up      |   z     | clockwise (nose swing to the right)

* Actual Results: 
axis    | label | direction of positive rotation
-----------------------------------------------------------
fwd     |  x     | right side DOWN
right   |   y     | nose down
up      |   z     | clockwise (nose swing to the right)

* Additional notes: These results are not even consistent with a proper right handed coordinate frame (in any orientation). Traditionally nose up means positive pitch. The graphic of the vehicle motion is fine, so it looks like the quirks have been compensated for internally.

digitalentity, 2018-04-17:
Essentially a duplicate of https://github.com/iNavFlight/inav/issues/50

closed 2018-04-17 by digitalentity

humanoid2050, 2018-04-17:
Almost a duplicate. My point is that the documentation does not match the actual implementation. #50 is just complaining about the choice of coordinate frames.