iNavFlight/inav #2993: Feature request: interuption and continuation of WP mission

wilco1967, 2018-03-31:
It would be nice if Waypoint missions could be interrupted, and then resumed.

Currently, if a waypoint mission is interrupted, and then re-started, the plane will go to the first waypoint, and start all over again from the beginning.

It would be nice if, once WP mission was at least initiated once during the flight, if I interrupt it, and then re-activate, it would go to the nearest waypoint it hasn't reached yet before. 

For example, if I start a waypoint mission, and it has flown already to WP 1 and 2, and is on it's way to WP3. Then I interrupt the mission, and fly around manually.
Now, If I re-enable waypoint mission, it should go the the nearest waypoint, it hasn't already reached previously. This could be WP3, if that happens to be closest, but also 5, or 10, or whatever is nearest, but NOT WP1 or 2.
If that is difficult, just going to the next WP (not necessary the closest) that hasn't been reached is also OK. As long as it doesn't start all the way from the beginning.

Many times when flying wapoint missions, I see something interesting, slightly off the mission path. Would be great if I could fly there manually, and then re-activate WP, and it would continue where it left off.... Can go in a straight line to the next WP... (don't have to fly back where we took manual control, although, that might be a nice option as well for others ?)

So far, I've flown many WP missions, the farthest out to 25 km. Works really well !

Also very interested in continuation of WP missions, when receiver signal lost, #2972 

Thanks !

access3424, 2018-11-13:
For sure this would be amazing!

timcable, 2021-11-19:
Exactly - I've been waiting for this idea to become reality.   Need to stop and take a few photos after reaching way-point # "X".  then be able to resume the route.

avsaase, 2021-11-19:
This is implemented in https://github.com/iNavFlight/inav/pull/6938.

closed 2021-11-19 by avsaase