iNavFlight/inav #2994: Only 1 Servo moves??!!

Raffapedia, 2018-04-01:
I have just received my Betaflight Omnibus F3 Pro from banggood, and I am trying to setup my nano talon airplane to work with INAV. So far I have flashed INAV 1.9 onto the board and gotten the gps to work, and the reciever tab shows inputs. When I use the customairplane to do mixing, no matter what servo number I put it always moves the same servo, WHICH MAKES NO SENSE!! This makes it impossible for me to make any other mix for the other servos. If anyone can help explain to me why any servo number I put on my mixer results in the same servo moving, that would be amazing.

JyeSmith, 2018-04-01:
Similar story here.  I've spent half the day trying to figure it out.

With Flying Wing mixer selected, servos 4 & 5 values change in the Motors tab. And only the servo attached to motor 3 pad moves.

With Airplane mixer selected, servos 4 & 5 values change in the Motors tab and both servos move.

This is with a lux_race FC, 1.9.1 build with fport and minimal setup.

shellixyz, 2018-04-01:
On OMNIBUS F3 the left servo needs to be connected to output 3 and the right one to output 4.
On LUX_RACE there is only one motor output the left servo needs to be connected to output 2 and the right one to output 3.

Raffapedia, 2018-04-01:
Alright Heres an update, I have been able to make a vtail mix that works, however the aileron servo in my nano talon will not move, no matter what kind of mix I do. I have changed it from pwm 5-8 and it will never move. I know for a fact the servo works since, if I plug it into pwm 3 or 4 it works, and I know for a fact that it is getting 5v of power so it is turned on. Does any one know why??

Here is the wiring diagram of my board
[image]

shellixyz, 2018-04-01:
I can't see your wiring diagram the picture is broken. If you are using the mini talon mixer from the wiki:

- The aileron servo should be connected to output 3 or 4
- The two v-tail servos should be connected to outputs 5 and 6

But note that you can't use the output 6 without removing the SBUS link and disabling UART3 so you can't use SBUS you need to use PPM for the receiver.

[image]

You should use this mix instead:
```
smix reset
smix 0 2 0 -100 0
smix 1 3 1 100 0
smix 2 4 1 -100 0
smix 3 3 2 -100 0
smix 4 4 2 -100 0
```

And connect the ailerons servo to output 3 and the v-tail servos to outputs 4 and 5. It should work without removing the link and would allow to use SBUS for the receiver.

JyeSmith, 2018-04-01:
@shellixyz thanks, I'll give that a go today.

Can I ask why the Lux is different and how do I find that info?

JyeSmith, 2018-04-01:
@shellixyz its works!

shellixyz, 2018-04-01:
@JyeSmith For some reason unknown to me only one motor output has been enabled for the Lux for fixed wing models. Normally you can find the info in the board docs [here](https://github.com/iNavFlight/inav/tree/master/docs) but it looks like there is no documentation for this board yet. Since there is no documentation you need to refer to the source code. You can see which output is assigned to what function in the [target.c](https://github.com/iNavFlight/inav/blob/master/src/main/target/LUX_RACE/target.c) file for the LUX_RACE target.

Raffapedia, 2018-04-02:
I have fixed it. I stopped using UART 3 on my board and switched my receiver to UART 2, and did what you recommended and the mix, and now I have been able to fix the plane, and its ready for its first maiden:)
Thanks.

closed 2018-04-04 by DzikuVx