iNavFlight/inav #2995: Add presets for 7-8 inch long range quads

artempolikarpov, 2018-04-01:
There is a such trend now 7-8 inch long range quads. I think it would be useful for many users to add special preset.

Motors are 2306, 2207 but with GPS, mini brushless gimbal or straight HD camera mount.

Now there are:
5 inch racer
5 inch GPS
10 inch standard quad

So it looks like gap for 7-8 inch quads.

DzikuVx, 2018-04-03:
@artempolikarpov if you can define such a preset in terms of PIDs and filtering, we can easily add it. I do not have such a platform to create a usable definition so some help will be required

Dronek, 2018-04-05:
It's rather questionable if pilots should be going long range if they rely on a preset to set up their quad. Apart from that I agree with @DzikuVx . Presets should be based on a stable tune. If you have a stable tune then suggest it to be used as a preset.

artempolikarpov, 2018-04-09:
[7inchGPSQuad.txt](https://github.com/iNavFlight/inav/files/1890389/7inchGPSQuad.txt)
Here is dump from 7inch gps quad. PIDs and filters could be taken from this dump

DzikuVx, 2018-04-10:
ok, saved. When time will allow then. In a mean time, move to configurator project:

iNavFlight/inav-configurator#404

DzikuVx, 2018-04-10:
Thannk @artempolikarpov

closed 2018-04-10 by DzikuVx