iNavFlight/inav #2996: Add automated efficiency mode and max current mode

jminix, 2018-04-01:
Add a new option in the navigation configuration that would enable the microcontroller to automatically seek the most effiecient flight speed in terms of amps/km.

This would be helpful to enable the drone cover the most ground possible and adapt to varying wind speeds.

Also, adding a max current setting, that governs the throttle to prevent it from going over a set number would be useful in preventing damage to a drive train system.

Both of these features would require accurately calibrated current meters and proper warning should be added in place. 

Have these things ever been tried before?  I am curious what some devs think.

shellixyz, 2018-04-01:
> Add a new option in the navigation configuration that would enable the microcontroller to automatically seek the most effiecient flight speed in terms of amps/km.

Not sure how easy that would be but I think it could be done. I don't think it has been tried before

> Also, adding a max current setting, that governs the throttle to prevent it from going over a set number would be useful in preventing damage to a drive train system.

Should be doable too. Ardupilot has this option.

Dronek, 2018-04-03:
There was also another discussion about a constant power mode somewhere in one of the many other issues here. I suggest to implement this before "max efficiency mode". The problem with latter one is: how should inav know whether max efficiency is reached or not? And what does it do if you fly at max efficiency and e.g. head wind increases? The efficiency will drop due to the wind but will this result in increasing throttle, decreasing or keeping it constant? This seems to include more airspeed functionality at least. Also "amps/km" does not represent efficiency. It should be rather Wh/km (or W/(km/h) if you like).

jminix, 2018-04-03:
Logically I don't think a max efficiency would be too hard to achieve.  Amps/km or Watts/km are known and accurate readings assuming you have correctly calibrated amp/voltage sensor and a valid GPS reading.  

Here is how I image it.  While navigating, INAV would (slowly) increase throttle and speed (something it already does) until the watts/km started to decrease.  It would then hold the last known GPS velocity corresponding to the highest efficiency.  It would then periodically test slightly faster and slightly slower GPS speeds and accept it if efficiency increases.  The readings would have to be averaged in 1 sec intervals or something like that.  Through testing it may be found that several seconds of average are needed to rule out short gusts of wind.  Perhaps if a significant heading change occurs (flying a different direction) the interval is reduced to account for the the change in wind direction more quickly.  This kind of feature would probably only be useful to someone flying on long range journeys like > 1 km.

I really don't know anything about INAV inner workings to do this my self and I can barely hobble through C programming, though when I get my new quad built I may try to do it if a dev has not done it yet.

shellixyz, 2018-04-03:
For the max efficiency I think the FC would need to "hunt" the best value raising/lowering the angle/throttle to find the peak constantly. Not sure how well this would work but it can be tested.

Energy calculation and Wh/km efficiency have already been added in 1.9.

shellixyz, 2018-06-03:
This issue should be left open

kf4mot, 2020-04-22:
My search for a way to set power by watts instead of a fixed value led me here. Beyond just being able to keep a consistent power setting my use case is evaluating power train efficiency in fixed wings.
Set power in Watts. Average several WP boxes or racetracks. Get Wh/km. Make changes and repeat as desired.

closed 2023-10-27 by DzikuVx