iNavFlight/inav #2997: Feature: osd Output for hd cam

JonnyFPV, 2018-04-02:
Hello, 

If it’s possible please make it possible to output the osd for hd video. 

Look at this https://youtu.be/jX2h8rJgNg8

In this Method the software puts every 1/2 second s png out. You can put it to you hd cam video later. 

Or look at parrot disco. You can do it there too.

Pairan, 2018-04-05:
I would like to have that but this is nothing the iNav fc code will have to do. I assume it to be something that can be made from the blackbox logs. 

If somebody can do such a program, step forward :)

fiam, 2018-04-05:
@Pairan is right. All you need from iNav is the blackbox log. All the rest is done by an external program.

shellixyz, 2018-04-05:
Yes that's what the blackbox_render program from the cleanflight blackbox tools does but it is not very flexible. It can only show gyro data, the sticks and some other things.

wolfpackt99, 2018-04-17:
https://youtu.be/VyY5opfCbWQ?t=305

More about the video transmitter i guess and what the iNav may spit out. For instance, in the video above, the Transmitter accepts two formats to overly over the video.

- mavlink
- vector open telemetry

my guess being you can't readily move about the osd bits though.

closed 2018-06-30 by stale[bot]