iNavFlight/inav #3001: Feature request: flight mode: stabilised climb and descent

craig34a, 2018-04-04:
It would be very handy to have a fixed climb or descent rate for cruise / stabilised flight or even position hold. 

This could be implemented as a setting within the inav app or even by way of a knob channel to change vertical speeds in flight. This feature would work best with an airspeed indicator to avoid a stall.

shellixyz, 2018-04-04:
You should already be able to do this with the `NAV ALTHOLD` flight mode and a mix in your remote control if it can do it.

craig34a, 2018-04-05:
Will test it and report back.

Dronek, 2018-04-05:
> That does not work. ALTHOLD is very solid and will not change altitude regardless of elevator...

You are actually wrong about this. Try pulling the elevator all the way back in ALTHOLD and the plane should be climbing. It will keep the new altitude whenever you let go off the elevator. If that doesn't work then something is off in your settings.

I suspect you're talking about fixed wing.

craig34a, 2018-06-12:
Thank you. You are right. It does work in Alt Hold.

closed 2018-06-12 by craig34a