iNavFlight/inav #3002: Feature request: Set direction of Position Hold

craig34a, 2018-04-04:
It would be handy to have an option to change the direction of position hold in flight between clockwise and counterclockwise wit a channel switch.

DzikuVx, 2018-04-12:
Please explain how this should be working and why? If it is only about switching left/right then modern radios can do it using mixer

craig34a, 2018-06-12:
Position hold in fixed wings results in a steady orbit around a point. This is clockwise right now. It would be nice to have the option to make this either clockwise or counter clockwise by default and have an option to change it in flight by using the rudder stick. This is how the Parrot Disco does it. a simple rudder movement can change the direction of the position hold orbit in fixed wings.

hali9, 2018-06-13:
@craig34a 
In file  `inav/src/main/navigation/navigation_fixedwing.c`
there is a function `static void calculateVirtualPositionTarget_FW(float trackingPeriod)`
and there is:
```
    // Calculate virtual position for straight movement
    if (needToCalculateCircularLoiter) {
        // We are closing in on a waypoint, calculate circular loiter
        float loiterAngle = atan2_approx(-posErrorY, -posErrorX) + DEGREES_TO_RADIANS(45.0f);
```

try change sign to negative, then compile and flash into FC:
`float loiterAngle = atan2_approx(-posErrorY, -posErrorX) - DEGREES_TO_RADIANS(45.0f);`
this should cause that plane will circling counterclockwise.

If this will work You can create a CLI variable to set default (left/right) direction of circling,
and then You can create a BOX in  `inav/src/main/fc/rc_modes` for changing this in fly.

giacomo892, 2018-08-12:
This can be interesting to have.. Not making this expire.