iNavFlight/inav #3003: Feature request: Emergency landing flight mode

brat002, 2018-04-04:
Would be nice to have an ability to trigger Emergency Landing flight mode for flying wing.

brat002, 2018-04-18:
No. I want to stay with normal RTL in case of Failsafe, but have ability to trigger land with ignorance of Signal lost.

fiam, 2018-04-18:
@brat002 `nav_rth_allow_landing = FS_ONLY` will do exactly that. We need to document it better though.

fiam, 2018-04-18:
Ah wait, you mean the opposite of what I thought of (land in manually triggered RTH, don't land on FS RTH). Why would you want to do that? If the RX cable breaks while in the air, the aircraft will continue flying until it keeps the battery and then fall like a rock.

Vereshchaginag, 2018-06-15:
Hi, agree with topic starter, we need emergency landing as a flight mode.
Last weekend I have flight with almost empty battery and no one possibility to move to home location on remaining power level. 
I have tried to move close to ground as possible, but if it was to close to the ground FS activated RTH procedure and quad moves up to 40 meters(as programmed). So I had to make disarm on approx 15 meters.
And if it was to possible to command to my quad emergency land it was be in one piece.
I think best realization of this mode is:
After mode enabled UAV should start lose attitude like usual landing procedure and SHOULD ignore any possible FS command ( it can appear, because UAV will be shadowed by trees or any other obstacles)
But due landing in unknown place, not at home position, and there can be high grass motors and ESC can be damaged, so disarm should be earlier than complete landing procedure.

giacomo892, 2018-06-15:
There is a little thing you are ignoring. How does the FC knows when disarm rather than it is stopping descending? Terrain altitude can be much different in the place you decide to land compared to home altitude. Rather than that, it will be good to have.

Vereshchaginag, 2018-06-15:
Maybe on sudden accelerometer or gyro changes, if prop will strike on any obstacle FC can feel that something wrong happens and then stop motors

closed 2018-08-28 by stale[bot]