iNavFlight/inav #3009: Deprecat feature SERVO_GIMBAL

DzikuVx, 2018-04-05:
Gimbal output can be performed via mixer without any additional featrures.
- [ ] Drop servo_gimbal feature
- [ ] Add gimbal type options to mixer inputs

markandkymward, 2018-04-06:
Can you point me to where this is described in detail?

shellixyz, 2018-04-06:
@markandkymward Where what is described in detail ? This is just a note about the current development. For the moment to use the servo gimbal you need to enable the SERVO_TILT feature. The first two servo outputs becomes the outputs for the tilt and pan servos.

markandkymward, 2018-04-06:
I have a MATEKF405-CTR and would like to use S5 or S6 to provide stabilized pitch to my pan/tilt.  I understand that INAV does not do resource remapping.  I know that this isn't a forum for this kind of assistance but without remapping I'm stuck.  Any way to point me in the right direction for working custom firmware?  I have the tool chain all set and am able to make basic changes to the firmware successfully.  I'm just stuck on how to get S5 to be recognized as a PWM our for the servo.

stronnag, 2018-04-07:
@markandkymward  Can you see if your requirement has been addressed in development with the new  MATEKF405_SERVOS6 target? There's a recent hex at  http://seyrsnys.myzen.co.uk/inav_dev/

markandkymward, 2018-04-07:
Stonnag to the rescue yet again.  I'll take a look back on the bench.

Again - I know that this isn't a support forum but would smix 0 5 1 100 0 work?  (S6, pitch stabilized output, 100% rate, 0 limit) 

Is there a chance I can get the target.c and target.h?  I would like to learn more about the internals of inav.   Comparing these files to the baseline would go a long way in increasing my understanding.

stronnag, 2018-04-07:
merely the facilitator. The real work was done in #2954

This is now an 'official' target (at least in the development branch). All the files, including updated board documentation are in the development branch (MATEKF405 target and some #define trickery). 
I'm no authority on smix, so I'll pass on that question.

PR #3187 (2018-05-09): Feature SERVO_TILT removed
Closes #3009 

SERVO_TILT can be performed using only Mixer, without any additional features. Docs update will follow
closed 2018-05-08 by DzikuVx