iNavFlight/inav #3010: Airspeed data not available for off-board OSDs (e.g. minimOSD / MWOSD)

Dronek, 2018-04-05:
### Board and Version
SP Racing F3 deluxe
INAV 1.9 on flying wing
INAV/SPRACINGF3 1.9.0 Mar  4 2018 / 23:41:56 (dc4ef594f)

### Behavior
- analog airspeed sensor assigned to RSSI ADC is working correctly when logged in blackbox
- problem: airspeed shows constant 0.0 km/h in MWOSD 1.8 running on a connected minimOSD
- assumption: airspeed value not available in MSP or not correctly read out by MWOSD ?

Is airspeed supported in on-board OSDs only? If so, please consider making it available at least for MWOSD which is a very popular OSD, works nicely with inav and as of now is more advanced than inav OSD in my opinion.

shellixyz, 2018-04-05:
The air speed will be available through MSP in the next version (#2853)

giacomo892, 2018-04-05:
@shellixyz May we find a place to report it even on LTM? Seems like we have some unused fields and it can be handy to be on light telemetry too

Dronek, 2018-04-05:
> The air speed will be available through MSP in the next version (#2853)

This sounds like a dream coming true after a year of waiting :)

shellixyz, 2018-04-05:
@giacomo892 It is already reported through LTM: https://github.com/iNavFlight/inav/blob/c66c28e7dc1e3c1ec8d1475f840691011ca712ba/src/main/telemetry/ltm.c#L192

stronnag, 2018-04-05:
in the extant 'S' frame ...

giacomo892, 2018-04-05:
my bad. didn't check the code carefully

Dronek, 2018-04-06:
https://github.com/ShikOfTheRa/scarab-osd/issues/504

Just for the sake of coherence.