iNavFlight/inav #3012: OSD not working on KFC32F3_INAV because wrong clock for MAX7456

marbalon, 2018-04-06:
After version 1.8 there is a lot of refactoring and new layers for bus SPI and i2c. But one thing is missing - changing clock before writing to specific device. On this target clock for OSD need to be STANDARD and this is set during initialization, but later there is a MPU6000 initialization and this change clock to FAST and OSD is not working anymore.

So in my opinion we have two solutions:
- remember clock for every device when it is inited, maybe by extending busDevice_t structure and then change it before write/read to device. But I don't know if every SPI device have own structure...
- or remember slowest clock for specific device and never increase it, but IHMO firs solution is more elegant.

Any other suggestion how to fix this problem?

digitalentity, 2018-04-16:
Yes, the first solution is more elegant. We need to look into it

marbalon, 2018-04-16:
The temporary simple solution can be #define for specified target eg. MAX_SPI_SPEED.

lgojniczek, 2018-07-06:
Is there any chance to include this in 2.0?

fiam, 2018-07-06:
@lgojniczek If someone submits a fix affecting only this target, it can be included in next RC.

shellixyz, 2018-07-06:
This could be related to the issue with the SDcard on other boards. For example the issue I have with the OmnibusF4PRO board. The SDcard and OSD chip are on the same SPI bus on this board.

lgojniczek, 2018-07-06:
> I will try t implement simple solution for now with MAX_SPI_SPEED for
> target - if defined for the board FC use it when to limit SPI speed.

Sounds good, thank you

fiam, 2018-07-11:
Fixed in #3532

closed 2018-07-11 by fiam