iNavFlight/inav #3013: SUPPORT FOR NEW SENSORS??

badandy1223, 2018-04-06:
Board and Version - ALL

Is this a new feature request? Yes

I was wondering if support for external sensors could be added. For example TEMPERATURE SENSORS and RPM SENSORS. These would be extremely helpful. FRSKY has the sensors, TEMS-01 and RPMS-01. These are examples of what I'm talking about. Useful for monitoring BATTERY TEMPS, MOTOR TEMPS, or MOTOR RPMS.

shellixyz, 2018-04-06:
I was thinking about adding support for SA56004 I2C temperature sensors. No sure how easy it would be to add support for the FrSky sensors. I guess you would need one more UART for connecting these.

badandy1223, 2018-04-06:
Yea, In most of my configurations, there's a spare UART not in use. Especially if using CRSF as the TX and RX on on the same UART port. That frees up one that used to be reserved for telemetry on SBUS...

All in all, I was just listing the FRSKY sensors as an example, not to say use these particular sensors. The SA56004 I2C temperature sensors look like they'll do the job, and by all means continue development on these. I would just like support to monitor battery temps,...no matter how this cat gets skinned.

badandy1223, 2018-04-06:
What about this sensor?

https://www.nxp.com/docs/en/data-sheet/LM75A.pdf

wx4cb, 2018-04-10:
FRSky sensors are all SPort, the protocol is already in the code for sending it out, i dunno how easy it would be for adding it as an input on the same uart that sport telemetry is on ?

I've actually just started playing with the sport interface on the sport gps etc that i have here, using the first link below:

Here's a couple links to docs:
https://www.ordinoscope.net/static/frsky-arduino/FrskySP/doc/html/index.html
https://code.google.com/archive/p/telemetry-convert/wikis/FrSkySPortProtocol.wiki

Denisfk1985, 2018-04-18:
I fully agree with the author. I would also like to be able to see in the OSD the data from an external thermometer. For example, the widespread onewire DS18b20, which you can connect a lot to one bus at a time. 
On my attached screen, the temperature is displayed from the temperature sensor built into the barometer. In the lower right corner (-53 degrees Celsius).
Sorry for the lack of pictures, flew at night.

[image]

closed 2018-07-01 by stale[bot]