iNavFlight/inav #3016: Oscillations through terrible Quadcopter Presets

Tavero, 2018-04-07:
Hello there,
Is there anyone who is using the 5" GPS 10"  or 12" Presets sucessfully? I've used them on my 450mm Quad and got terrible oscillations from the beginning. The Oscillations in the 10" Preset were so bad, it actually broke an arm during idle throttle. These issues are nothing new https://github.com/iNavFlight/inav/issues/2605.  In my opinion the pids are way too agressive and probably only worked for old generation gyros.  In that case it would be better to remove these presets completely.

[LOG00010.TXT](https://github.com/iNavFlight/inav/files/1886674/LOG00010.TXT)
[LOG00011.TXT](https://github.com/iNavFlight/inav/files/1886675/LOG00011.TXT)
As you can see in my logs the PIDs were oscillation the Motors and gyros into an urecoverable vicious cycle. Gyro is working fine. Motors and props are balanced and running smoothly.

greetings Tav

Antiheld86, 2018-04-09:
I had the same experience with 10" preset on a 420mm quad.

Tavero, 2018-04-09:
Hi there Antiheld,
were you able to solve this issue? After 3 days of frustration i still cant get a stable flight.
THIS ist the best I could get. And it used default settings. https://youtu.be/vq6BqcRrCsM
[LOG00018.TXT](https://github.com/iNavFlight/inav/files/1891278/LOG00018.TXT)
It still suffers from self inflicted oscillations.

Changing the PIDs made it worse. No idea what else I should do

digitalentity, 2018-04-10:
The problem with big quads is that you can only create preset for a very specific setup. There is a bug variety of motors and ESCs, with/without braking, 3S/4S, different frame rigidity etc. Too many factors to consider for making a perfect preset.

Antiheld86, 2018-04-10:
Hi @Tavero ,
Are you in Level mode or in PH? I had the biggest oscillations in PH. To get it more stable, I used the default settings and started to tweak the pid's from there. It worked much better but still not perfect. But i had not much time the last days to try more. I had to solve differnt problems like fixing one arm wich was a bit broken because of disarming in the air and very short rx range because of faulty R8X. I hope I can try in the next days.

I got better results by lowering the D value of the GPS Nav PID's.

@digitalentity I think you are right that it is difficult to make a preset. But probably it would be better not having a preset, than having one with the big chance of worse flight behavior than with default settings. My quad is exactly within the specs that the 10"-preset says it is for. But it is not flyable with it.

Tavero, 2018-04-11:
Hey,
The answer is everywhere. It began in Level and Acro mode.  Couldn't even get it in the air.
I reduced P and increased D. It is now flying stable with 35/30/35 in roll and pitch.

it is also oscillating around a gps point in PS.  Stops oscillating if it its drifting further away from it

sincoder, 2018-04-15:
@Tavero also reduce NAV PIDs