iNavFlight/inav #3017: Inav 1.8 little or no movement of servos

bmoreking443, 2018-04-07:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.
Hello all,
Currently working on my 3rd inav build and this is where the problem lies. I'm using a omnibus F4 pro v2, digital 9g servos , matek pdb. Servos are powered from a 5v pad on the pdb .Currently I can arm my plane but when I increase the throttle servo 3 moves with the throttle stick movement which I know is wrong. My motor is connected to pwm 1 and does spin in the motors tab. When rolling left servo 3 on pwm 3 moves correctly. When rolling right servo 4 connected to pwm 4 moves very little if at all. Servos have been tested and they work fine. I can get full range of servo 4 when using the throttle stick which is weird. I've changed the board and pdb and the issue followed. Can you explain or give a little insight on how I can resolve the servo issue.
### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

shellixyz, 2018-04-07:
Please post a config dump

bmoreking443, 2018-04-08:
# version
# INAV/OMNIBUSF4PRO 1.8.0 Nov  1 2017 / 06:34:32 (912d1315)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 -100 3 
servo 4 1000 2000 1500 100 4 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature GPS
feature LED_STRIP
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
[... 474 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set tpa_rate = 33
set tpa_breakpoint = 1300

bmoreking443, 2018-04-08:
thx for the help on this just to let you know this is the second f4 board with exactly the same issue.

shellixyz, 2018-04-08:
I think your issue is because you enabled channel forwarding for servo 3 and 4. In the servo tab uncheck all the checkboxes.

bmoreking443, 2018-04-08:
ok i will try un-checking  the servo boxes. After watching the youtube link above I understand the servo movement a bit more but, now i guess my question would be if I want full servo movement on the bench do i need to flash a earlier version of Inav? Maybe 1.71 or 1.73 also i may have failed to mention before that i am using a simonK esc  does Inav behave like betaflight where as you need to use a BLheli esc? If so i can change that out.

shellixyz, 2018-04-08:
You don't need an earlier version. If the flying wing mixer have been changed it is because it was wrong before. Also iNav should work fine with SimonK ESCs.

shellixyz, 2018-04-08:
Yes you can paste the configuration from the board in the working plane in the CLI of the non working one. Just don't forget to change the board orientation appropriately and redo the accelerometer/voltage/current calibration afterwards. But you shouldn't have to do that to setup a wing it should be easy.

- Set the mixer to flying wing
- Adjust the board orientation to match the way you installed the board and the angle of attack you need
- Reverse the servo(s) as needed
- Adjust the servos middle so that your elevons have the needed reflex and the servos min/max
- Set the pitch and roll rates to your wing maximum rates and the pitch and roll FF factor
- Lower the pitch and roll rates if you want them lower
- Setup your flight modes

Nothing else needs to be done for MANUAL/ACRO/ANGLE/HORIZ. Go fly and tune the P and I for roll and pitch.

closed 2018-04-12 by DzikuVx