iNavFlight/inav #3019: Can MSPv2 still read MSPv1 messages?

benparsonage95, 2018-04-08:
I have a device that transmits GPS waypoint data using MSPv1, intended to be used in conjunction with GCS_NAV to allow 'follow me' functionality (as demonstrated/developed by iforce2d for iNav 1.2/1.3 https://www.youtube.com/watch?v=fA5sFRh3OkY&lc=z22wfxm4ds3bsjhd2acdp432swbz4txfukpamkbg4a1w03c010c). 

Do I need to modify the code for this device to transmit in MSPv2 instead or can iNav still handle MSPv1? 

Thank you.

stronnag, 2018-04-08:
handles both, replies in the format it was requested.

digitalentity, 2018-04-10:
You can use MSPv1, it's supported. New 16-bit MSP request IDs will only work with MSPv2 (MSPv1 has only 8-bit reqeust ID field).

closed 2018-04-10 by digitalentity