iNavFlight/inav #3020: Matek F722-STD - with Inav 1.9.1 - FC disoriented in the air

DaveKraft, 2018-04-08:
I just finish my first Inav build, it is a cloudsurfer from HK with Matek F722 FC, and Inav 1.9.1 (recommended version from mateksys, with fixed issue of rudder output -S06). On the ground it looks just fine, GPS is fixed, just ready to go, but once it airborne, the flight controller looks disoriented, look at the video, how horizont bar starts drifts all over screen in completely wrong angle related to ground. 
then I make some changes:
I tried to disable Barometric sensor (make it simple as I can), set lover LPF gyro cutoff freq (20MHz), I change how is FC mounted (add soft mount to my FC), balance prop again, and result is still the same, after takeoff it is works just fine, but after lets say 20 - 30sec horizont bar starts moving randomly all over screen, so FC is disoriented and any of flight mods can't be used, except "manual" mode.
Unfortunately, I don't have black box data. Only a video. And I can't reproduce same behavior on the ground, I hold airplane in hand and wiggle with for 10minutes with motor on cruise throttle and FC was working just fine, issue appears only in the air, during actual flight.

Video link:
https://youtu.be/3vWMR8YK-SE

my settings are:
# version
# INAV/MATEKF722 1.9.1 Mar  8 2018 / 11:09:25 (a818e2e)

# resources

# mixer
mixer AIRPLANE

# servo mix
smix 0 7 8 100 0

# servo
servo 2 1000 2000 1382 100 -1 
servo 3 1000 2000 1522 -100 -1 
servo 4 1000 2000 1522 -100 -1 
servo 5 1000 2000 1496 -100 -1 

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature CHANNEL_FORWARDING
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER5678

# name
name CloudSurfer

# serial
serial 0 1024 115200 38400 0 115200
serial 2 32 115200 38400 0 115200
serial 3 2 115200 38400 0 115200
serial 4 2048 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 7 1700 2100
aux 1 1 8 1300 1700
aux 2 2 8 1700 2100
aux 3 3 1 1475 1900
aux 4 9 1 1675 1900
aux 5 8 1 1950 2100
aux 6 10 1 1125 1475
aux 7 14 1 900 1025
aux 8 27 6 1700 2100
aux 9 28 6 900 1300
aux 10 15 2 1300 2100
aux 11 30 3 1300 1700

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 20HZ
set acc_hardware = MPU6500
set acczero_x = -74
set acczero_y = -23
set acczero_z = -103
set accgain_x = 4087
set accgain_y = 4085
set accgain_z = 4088
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set min_throttle = 1275
set max_throttle = 2000
set failsafe_throttle = 1350
set failsafe_procedure = RTH
set align_board_pitch = -30
set align_board_yaw = 1800
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 310
set battery_capacity = 6000
set battery_capacity_warning = 3000
set battery_capacity_critical = 1200
set current_meter_scale = 179
set imu_dcm_kp = 2108
set imu_dcm_ki = 16384
set imu_dcm_kp_mag = 2048
set imu_dcm_ki_mag = 16384
set small_angle = 32
set auto_disarm_delay = 0
set gps_sbas_mode = EGNOS
set nav_extra_arming_safety = OFF
set nav_landing_speed = 150
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 7000
set nav_fw_cruise_thr = 1600
set nav_fw_min_thr = 1275
set nav_fw_max_thr = 2000
set nav_fw_bank_angle = 35
set nav_fw_pitch2thr = 12
set nav_fw_loiter_radius = 7000
set telemetry_inversion = ON
set osd_video_system = 1
set osd_rssi_alarm = 35
set osd_time_alarm = 60
set osd_alt_alarm = 1000
set osd_main_voltage_pos = 289
set osd_rssi_pos = 2050
set osd_flymode_pos = 2476
set osd_throttle_pos = 193
set osd_vtx_channel_pos = 2487
set osd_current_draw_pos = 2401
set osd_mah_drawn_pos = 353
set osd_bat_remaining_capacity_pos = 2433
set osd_gps_speed_pos = 2275
set osd_gps_sats_pos = 2081
set osd_gps_lon_pos = 2098
set osd_gps_lat_pos = 2066
set osd_home_dir_pos = 2092
set osd_home_dist_pos = 2059
set osd_altitude_pos = 2297
set osd_vario_pos = 2231
set osd_ontime_flytime_pos = 2455
set osd_messages_pos = 2497
set osd_main_cell_voltage_pos = 2369
set osd_throttle_auto_thr_pos = 2113
set osd_efficiency_mah_pos = 2465

# profile
profile 1

set fw_p_pitch = 10
set fw_i_pitch = 6
set fw_p_roll = 10
set fw_i_roll = 6
set fw_p_yaw = 10
set fw_i_yaw = 6
set fw_ff_yaw = 50
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set tpa_rate = 40
set pitch_rate = 15
set yaw_rate = 10
set manual_roll_rate = 80
set manual_pitch_rate = 80
set manual_yaw_rate = 80

shellixyz, 2018-04-08:
This doesn't seem right. Did you change these values ?

```
set imu_dcm_kp = 2108
set imu_dcm_ki = 16384
```

You could try to set them back to the defaults:

```
set imu_dcm_kp = 2500
set imu_dcm_ki = 50
```

There is known memory corruption bug on MATEKF722 board with version 1.9.0 so that might be be reason (#2914). I think it is not recommended to use this board with version 1.9.0.

If you feel adventurous you can try the development branch but use the default `imu_dcm_kp` and `imu_dcm_ki` values: [inav_1.9.1_MATEKF722.hex.zip](https://github.com/iNavFlight/inav/files/1887792/inav_1.9.1_MATEKF722.hex.zip)

Remember this is not an official release. It has been built from the latest code and has not been tested as well.

DaveKraft, 2018-04-08:
I did not change these values, to be honest, I don't even know what they are!! :-) I chose 1.9.1 (from mateksys page) simply because my board doesn't have support in older versions, is there any way how to use this board with older versions? like 1.8.0? But I will definitely set those values as you suggested, and will try the latest build (from your post). 

thx for your help

Regards, 
David

wilco1967, 2018-04-08:
The Matek F722 is working just fine in the latest Development... But not in the 'official' 1.9.0.
Those values being corrupted in 1.9.0 is already solved in Development. 

I've been flying the Development version on a Mini Talon the last few weeks or so, since that bug was fixed, and it's working great since.
Never tried flying while the bug was still there...

[inav_1.9.1_MATEKF722 03-04.hex.zip](https://github.com/iNavFlight/inav/files/1887887/inav_1.9.1_MATEKF722.03-04.hex.zip)

This is the version from a few days ago that is working great for me, and is 'flight tested'.... 
shellxyz's version seems even newer, so you may want to use that instead.

Blackbox logging however doesn't work properly yet (at least on mine), but that's not stopping it from flying. #2796

shellixyz, 2018-04-08:
@DaveKraft The support for the MATEKF722 board has only be added recently and the first release including it was 1.9.0 so you can't use 1.8.0. I would suggest to use either my build (I flight tested it so it should work) or the build provided by @wilco1967.

DaveKraft, 2018-04-08:
Thx for your help guys. I already flash version from shellxyz and set everything from scratch, these "strange" values are no longer there, so I will test it as soon as I can, probably next weekend.

One thing I didn't mentioned (I forgot), on my previous version (the faulty one) I was unable to save small_angle value (after reboot it was set back to 32, every time), don't know if it could be some way related to my issue, now I can save this value.

shellixyz, 2018-04-08:
> One thing I didn't mentioned (I forgot), on my previous version (the faulty one) I was unable to save small_angle value (after reboot it was set back to 32, every time), don't know if it could be some way related to my issue, now I can save this value.

Was most likely related to the same bug that caused the corruption of your `imu_dcm_kp` and `imu_dcm_ki` settings

DaveKraft, 2018-04-14:
Hi guys, I'm just come back from the airfield, and can confirm, that now it flew like a charm! :-) I tried almost every flight mods, let the plane on Pos/alt hold for more than 15minutes, angle, horizont and RTH is also working......everything is OK, I wasn't brave enough to test auto launch, maybe next time. :-)    Thank you again for your support.   
btw, I found only one thing strange, on previous build, I had to invert telemetry (frsky receiver) by CLI command, on this build telemetry work only if I disable/turn off telemetry inverter. everything is connected same way as before, don't know if it is a bug, or some change in this build, but anyway it works that way! :-)

here is my current and working settings:  
# version
# INAV/MATEKF722 1.9.1 Apr  8 2018 / 20:57:06 (2d3fa62b)

# resources

# mixer
mixer AIRPLANE

# servo mix
smix 0 7 8 100 0

# servo
servo 2 1000 2000 1382 100 -1 
servo 3 1000 2000 1522 -100 -1 
servo 4 1000 2000 1522 -100 -1 
servo 5 1000 2000 1496 -100 -1 

# feature
feature -TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature CHANNEL_FORWARDING
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map

# name
name CloudSurfer

# serial
serial 0 1024 115200 38400 0 115200
serial 2 32 115200 38400 0 115200
serial 3 2 115200 38400 0 115200
serial 4 2048 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 7 1700 2100
aux 1 1 8 1300 1700
aux 2 2 8 1700 2100
aux 3 3 1 1475 1900
aux 4 9 1 1675 1900
aux 5 8 1 1950 2100
aux 6 10 1 1125 1475
aux 7 14 1 900 1025
aux 8 27 6 1700 2100
aux 9 28 6 900 1300
aux 10 15 2 1300 2100
aux 11 30 3 1300 1700

# adjrange

# rxrange

# master
set looptime = 2000
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6500
set acczero_x = -62
set acczero_y = -30
set acczero_z = -39
set accgain_x = 4087
set accgain_y = 4087
set accgain_z = 4071
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 16
set min_throttle = 1275
set max_throttle = 2000
set failsafe_throttle = 1450
set failsafe_procedure = RTH
set align_board_pitch = -25
set align_board_yaw = 1800
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 310
set battery_capacity = 6000
set battery_capacity_warning = 3000
set battery_capacity_critical = 1200
set current_meter_scale = 179
set small_angle = 180
set auto_disarm_delay = 0
set gps_sbas_mode = EGNOS
set nav_extra_arming_safety = OFF
set nav_landing_speed = 150
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 7000
set nav_fw_cruise_thr = 1600
set nav_fw_min_thr = 1275
set nav_fw_max_thr = 2000
set nav_fw_bank_angle = 35
set nav_fw_pitch2thr = 12
set nav_fw_loiter_radius = 7000
set nav_fw_launch_thr = 2000
set nav_fw_launch_motor_delay = 250
set nav_fw_launch_climb_angle = 20
set telemetry_inversion = OFF
set osd_video_system = 1
set osd_rssi_alarm = 35
set osd_time_alarm = 60
set osd_alt_alarm = 1000
set osd_main_voltage_pos = 289
set osd_rssi_pos = 2050
set osd_flymode_pos = 2476
set osd_throttle_pos = 193
set osd_vtx_channel_pos = 2487
set osd_current_draw_pos = 2401
set osd_mah_drawn_pos = 353
set osd_bat_remaining_capacity_pos = 2433
set osd_gps_speed_pos = 2275
set osd_gps_sats_pos = 2081
set osd_gps_lon_pos = 2098
set osd_gps_lat_pos = 2066
set osd_home_dir_pos = 2092
set osd_home_dist_pos = 2059
set osd_altitude_pos = 2297
set osd_vario_pos = 2231
set osd_ontime_flytime_pos = 2455
set osd_messages_pos = 2497
set osd_main_cell_voltage_pos = 2369
set osd_throttle_auto_thr_pos = 2113
set osd_efficiency_mah_pos = 2465

# profile
profile 1

set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set manual_roll_rate = 70
set manual_pitch_rate = 70
set manual_yaw_rate = 70

DaveKraft, 2018-04-15:
well, I tested it again, and same behavior is back. but after much longer time than on the previous build. Today I flew more than 25 minutes, and after 20 minutes of climbing to higher altitude horizont start floating again, but much slower and not so random, just "waving" from one side to another, very slowly, but I wasn't able to use any automatic flight mod, because of that. my best guess is that my FC is getting overheating during fly , which leads to disoriented gyro/acelerometer, and the previous bug just amplified that same behavior.I will try to add some better cooling to my FC, and make few more tests.

putimir, 2018-06-04:
Hey @shellixyz : do you know any other parameters that get corrupted in 1.9.0? I flew my S800 yesterday (722/1.9.0) without much issues (it was maiden and used autolaunch haha; horizon, acro/air), everything was fine except the RTH, which exhibited violent banks and yanks into ground. Then I stumbled upon this issue (I also had corrupted set imu_dcm_kp and set imu_dcm_ki); I flashed the board yesterday with hex from Matek site (dated apr.27), but did not do full chip erase).

So, are there any other parameters known to check for corruption?

Thanks

closed 2018-08-17 by stale[bot]