iNavFlight/inav #3021: 1.9 RTH SPIRAL..TWO BOARDS TWO SHIPS

biomedchad, 2018-04-09:
### Betflight f3 and clraving f4air v2

### Behavior
upon first sat lock and launch the ships will spiral into the ground instead of rth.  after they are brought back and repowered they run just fine.  gps coordinates seem to be 180 off upon initial gps lock and launch. after reboot, they are normal and read fine.

_Is this a new feature request? NO

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

stronnag, 2018-04-09:
Please verify the nav sensors are working correctly by checking against https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup and in particular https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup#initial-flight-tests

Once these are complete, please provide a blackbox log, otherwise it is  unlikely that we can help you.

biomedchad, 2018-04-09:
I understand.  I believe this is now a heading issue..gps..not sure..same gps m8n mini that many use from 2 different vendors.  heading goes crazy on launch and then when I land and go into osd and reboot it works fine..hope that helps.   posting video on the inav facebook group

biomedchad, 2018-04-09:
https://www.youtube.com/watch?v=uD2rPs9HI7s

wx4cb, 2018-04-10:
are you using the gps AND a compass? try turning the compass off and just use the heading from the gps?

JDFPV, 2018-04-10:
Im having the very same issuse. And im not using a compass on my plane. Running omnibus f4 pro board. Ublock 8m gps. Everything worked perfectly fine on 1.8 firmware. Swithced to 1.9 and the problem started.

shellixyz, 2018-04-10:
Please provide a config dump and blackbox log.

JDFPV, 2018-04-10:
Here's my dump files. I did not have a sd on board when this happened to me using rth. But this issues has happened to me twice. 

# dump

# version
# INAV/OMNIBUSF4PRO 1.9.0 Mar 27 2018 / 03:57:42 (dc4ef594)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 -100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
beeper LAUNCH_MODE
beeper CAM_CONNECTION_OPEN
beeper CAM_CONNECTION_CLOSED

# map
map AETR5678

# name
name -

# serial
serial 20 1 115200 38400 0 115200
serial 0 64 115200 38400 0 115200
serial 2 64 115200 38400 0 115200
serial 5 2 115200 115200 0 115200

# led
led 0 0,0::C:0
led 1 0,0::C:0
led 2 0,0::C:0
led 3 0,0::C:0
led 4 0,0::C:0
led 5 0,0::C:0
led 6 0,0::C:0
led 7 0,0::C:0
led 8 0,0::C:0
led 9 0,0::C:0
led 10 0,0::C:0
led 11 0,0::C:0
led 12 0,0::C:0
led 13 0,0::C:0
led 14 0,0::C:0
led 15 0,0::C:0
led 16 0,0::C:0
led 17 0,0::C:0
led 18 0,0::C:0
led 19 0,0::C:0
led 20 0,0::C:0
led 21 0,0::C:0
led 22 0,0::C:0

JDFPV, 2018-04-10:
led 23 0,0::C:0
led 24 0,0::C:0
led 25 0,0::C:0
led 26 0,0::C:0
led 27 0,0::C:0
led 28 0,0::C:0
led 29 0,0::C:0
led 30 0,0::C:0
led 31 0,0::C:0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

# mode_color
mode_color 0 0 1
mode_color 0 1 11
mode_color 0 2 2
mode_color 0 3 13
mode_color 0 4 10
mode_color 0 5 3
mode_color 1 0 5
mode_color 1 1 11
mode_color 1 2 3
mode_color 1 3 13
mode_color 1 4 10
mode_color 1 5 3
mode_color 2 0 10
mode_color 2 1 11
mode_color 2 2 4
mode_color 2 3 13
mode_color 2 4 10
mode_color 2 5 3
mode_color 3 0 8
mode_color 3 1 11
mode_color 3 2 4
mode_color 3 3 13
mode_color 3 4 10
mode_color 3 5 3
mode_color 4 0 7
mode_color 4 1 11
mode_color 4 2 3
mode_color 4 3 13
mode_color 4 4 10
mode_color 4 5 3
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3
mode_color 6 0 6
mode_color 6 1 10
mode_color 6 2 1
mode_color 6 3 0
mode_color 6 4 0
mode_color 6 5 2
mode_color 6 6 3
mode_color 6 7 6
mode_color 6 8 0
mode_color 6 9 0
mode_color 6 10 0

# aux
aux 0 0 3 1200 1325
aux 1 1 0 1450 1575
aux 2 2 0 1025 1150
aux 3 3 0 1400 1600
aux 4 8 1 1850 2025
aux 5 19 1 1000 1175
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# master
set looptime = 2000
set gyro_sync = ON
set gyro_sync_denom = 2

JDFPV, 2018-04-10:
set align_gyro = DEFAULT
set gyro_hardware_lpf = 20HZ
set gyro_lpf_hz = 60
set moron_threshold = 32
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 1
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 1
set vbat_adc_channel = 2
set rssi_adc_channel = 3
set current_adc_channel = 1
set airspeed_adc_channel = 0
set acc_notch_hz = 0
set acc_notch_cutoff = 1
set align_acc = DEFAULT
set acc_hardware = MPU6000
set acc_lpf_hz = 15
set acczero_x = 144
set acczero_y = -66
set acczero_z = -259
set accgain_x = 4081
set accgain_y = 4068
set accgain_z = 4040
set rangefinder_hardware = NONE
set rangefinder_median_filter = OFF
set align_mag = DEFAULT
set mag_hardware = NONE
set mag_declination = 0
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0
set mag_calibration_time = 30
set baro_hardware = NONE
set baro_median_filter = ON
set pitot_hardware = NONE
set pitot_use_median_filter = ON
set pitot_noise_lpf =  0.600
set pitot_scale =  1.000
set receiver_type = SERIAL
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 100
set rssi_invert = OFF
set rc_smoothing = ON
set serialrx_provider = SBUS
set sbus_inversion = ON
set rx_spi_rf_channel_count = 0
set spektrum_sat_bind = 0
set rx_min_usec = 885
set rx_max_usec = 2115
set serialrx_halfduplex = OFF
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SDCARD
set sdcard_detect_inverted = ON
set min_throttle = 1000
set max_throttle = 2000
set min_command = 1000
set motor_pwm_rate = 400
set motor_pwm_protocol = STANDARD
set failsafe_delay = 5
set failsafe_recovery_delay = 5
set failsafe_off_delay = 200
set failsafe_throttle = 1000
set failsafe_throttle_low_delay = 100
set failsafe_procedure = RTH
set failsafe_stick_threshold = 50
set failsafe_fw_roll_angle = -200
set failsafe_fw_pitch_angle = 100
set failsafe_fw_yaw_rate = -45
set failsafe_min_distance = 0
set failsafe_min_distance_procedure = DROP
set align_board_roll = 22
set align_board_pitch = -40
set align_board_yaw = 0
set gimbal_mode = NORMAL
set vbat_scale = 1100
set vbat_max_cell_voltage = 430
set vbat_min_cell_voltage = 200
set vbat_warning_cell_voltage = 250
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH
set current_meter_scale = 700
set current_meter_offset = 0
set current_meter_type = ADC
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set servo_center_pulse = 1500
set servo_pwm_rate = 60
set servo_lpf_hz = 20
set flaperon_throw_offset = 200
set tri_unarmed_servo = ON
set reboot_character = 82
set imu_dcm_kp = 2500
set imu_dcm_ki = 50
set imu_dcm_kp_mag = 10000
set imu_dcm_ki_mag = 0
set small_angle = 180
set fixed_wing_auto_arm = OFF
set disarm_kill_switch = OFF
set auto_disarm_delay = 60
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_dyn_model = AIR_1G
set gps_auto_config = ON
set gps_auto_baud = ON
set gps_ublox_use_galileo = OFF
set gps_min_sats = 6
set deadband = 5
set yaw_deadband = 5
set pos_hold_deadband = 20
set alt_hold_deadband = 50

biomedchad, 2018-04-10:
It’s all in my video. No compass or mag. All turned off. This is the same fixed wing setup that everyone uses and the wiki recommended.

shellixyz, 2018-04-10:
@biomedchad I can't see any RTH in your video and a video is not the same as a flight log.

shellixyz, 2018-04-10:
@JDFPV You config is incomplete and please don't post in multiple comment.

JDFPV, 2018-04-10:
set 3d_deadband_throttle = 50
set fw_autotune_overshoot_time = 100
set fw_autotune_undershoot_time = 200
set fw_autotune_threshold = 50
set fw_autotune_ff_to_p_gain = 10
set fw_autotune_ff_to_i_tc = 600
set inav_auto_mag_decl = ON
set inav_gravity_cal_tolerance = 5
set inav_use_gps_velned = ON
set inav_reset_altitude = FIRST_ARM
set inav_max_surface_altitude = 200
set inav_w_z_surface_p =  3.500
set inav_w_z_surface_v =  6.100
set inav_w_z_baro_p =  0.350
set inav_w_z_gps_p =  0.200
set inav_w_z_gps_v =  0.500
set inav_w_xy_gps_p =  1.000
set inav_w_xy_gps_v =  2.000
set inav_w_z_res_v =  0.500
set inav_w_xy_res_v =  0.500
set inav_w_acc_bias =  0.010
set inav_max_eph_epv =  1000.000
set inav_baro_epv =  100.000
set nav_disarm_on_landing = OFF
set nav_use_midthr_for_althold = OFF
set nav_extra_arming_safety = ON
set nav_user_control_mode = ATTI
set nav_position_timeout = 5
set nav_wp_radius = 100
set nav_wp_safe_distance = 10000
set nav_auto_speed = 300
set nav_auto_climb_rate = 500
set nav_manual_speed = 500
set nav_manual_climb_rate = 200
set nav_landing_speed = 200
set nav_land_slowdown_minalt = 500
set nav_land_slowdown_maxalt = 2000
set nav_emerg_landing_speed = 500
set nav_min_rth_distance = 500
set nav_rth_climb_first = ON
set nav_rth_climb_ignore_emerg = OFF
set nav_rth_tail_first = OFF
set nav_rth_allow_landing = NEVER
set nav_rth_alt_mode = AT_LEAST
set nav_rth_abort_threshold = 50000
set nav_rth_altitude = 15240
set nav_mc_bank_angle = 30
set nav_mc_hover_thr = 1500
set nav_mc_auto_disarm_delay = 2000
set nav_fw_cruise_thr = 1500
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1800
set nav_fw_bank_angle = 35
set nav_fw_climb_angle = 20
set nav_fw_dive_angle = 15
set nav_fw_pitch2thr = 10
set nav_fw_loiter_radius = 4000
set nav_fw_land_dive_angle = 2
set nav_fw_launch_velocity = 300
set nav_fw_launch_accel = 1863
set nav_fw_launch_max_angle = 45
set nav_fw_launch_detect_time = 40
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set nav_fw_launch_motor_delay = 500
set nav_fw_launch_spinup_time = 100
set nav_fw_launch_min_time = 0
set nav_fw_launch_timeout = 5000
set nav_fw_launch_max_altitude = 0
set nav_fw_launch_climb_angle = 18
set telemetry_switch = OFF
set telemetry_inversion = OFF
set frsky_default_latitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = METRIC
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_sound_interval = 5
set smartport_uart_unidir = OFF
set smartport_fuel_unit = MAH
set ibus_telemetry_type = 0
set ltm_update_rate = NORMAL
set ledstrip_visual_beeper = OFF
set osd_video_system = 0
set osd_row_shiftdown = 0
set osd_units = IMPERIAL
set osd_stats_energy_unit = MAH
set osd_rssi_alarm = 20
set osd_time_alarm = 200
set osd_alt_alarm = 1524
set osd_artificial_horizon_reverse_roll = OFF
set osd_crosshairs_style = DEFAULT
set osd_left_sidebar_scroll = NONE
set osd_right_sidebar_scroll = NONE
set osd_sidebar_scroll_arrows = OFF
set osd_main_voltage_pos = 2456
set osd_main_voltage_decimals = 1
set osd_rssi_pos = 23
set osd_flytimer_pos = 2071
set osd_ontime_pos = 279
set osd_flymode_pos = 2444
set osd_throttle_pos = 65
set osd_vtx_channel_pos = 200
set osd_crosshairs_pos = 0
set osd_artificial_horizon_pos = 2248
set osd_current_draw_pos = 2425
set osd_mah_drawn_pos = 2401
set osd_wh_drawn_pos = 161
set osd_bat_remaining_capacity_pos = 193
set osd_bat_remaining_percent_pos = 225
set osd_craft_name_pos = 84
set osd_gps_speed_pos = 2241
set osd_gps_sats_pos = 2049
set osd_gps_lon_pos = 402
set osd_gps_lat_pos = 384
set osd_home_dir_pos = 2126

JDFPV, 2018-04-10:
set osd_home_dist_pos = 2433
set osd_altitude_pos = 2264
set osd_vario_pos = 183
set osd_vario_num_pos = 248
set osd_pid_roll_pos = 322
set osd_pid_pitch_pos = 354
set osd_pid_yaw_pos = 386
set osd_power_pos = 47
set osd_air_speed_pos = 163
set osd_ontime_flytime_pos = 24
set osd_rtc_time_pos = 407
set osd_messages_pos = 2338
set osd_gps_hdop_pos = 320
set osd_main_cell_voltage_pos = 44
set osd_throttle_auto_thr_pos = 70
set osd_heading_graph_pos = 2058
set osd_efficiency_mah_pos = 161
set osd_efficiency_wh_pos = 161
set i2c_speed = 400KHZ
set debug_mode = NONE
set acc_task_frequency = 500
set attitude_task_frequency = 250
set async_mode = NONE
set throttle_tilt_comp_str = 0
set input_filtering_mode = OFF
set mode_range_logic_operator = OR
set stats = OFF
set stats_total_time = 0
set stats_total_dist = 0
set tz_offset = 0
set display_force_sw_blink = OFF

# profile
profile 1

set mc_p_pitch = 40
set mc_i_pitch = 30
set mc_d_pitch = 23
set mc_p_roll = 40
set mc_i_roll = 30
set mc_d_roll = 23
set mc_p_yaw = 85
set mc_i_yaw = 45
set mc_d_yaw = 0
set mc_p_level = 20
set mc_i_level = 15
set mc_d_level = 75
set fw_p_pitch = 2
set fw_i_pitch = 15
set fw_ff_pitch = 70
set fw_p_roll = 2
set fw_i_roll = 15
set fw_ff_roll = 30
set fw_p_yaw = 50
set fw_i_yaw = 45
set fw_ff_yaw = 0
set fw_p_level = 10
set fw_i_level = 15
set fw_d_level = 75
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set dterm_setpoint_weight =  0.000
set fw_iterm_throw_limit = 165
set fw_reference_airspeed =  1000.000
set fw_turn_assist_yaw_gain =  1.000
set dterm_notch_hz = 0
set dterm_notch_cutoff = 1
set pidsum_limit = 500
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000
set heading_hold_rate_limit = 90
set nav_mc_pos_z_p = 50
set nav_mc_pos_z_i = 0
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 40
set pitch_rate = 15
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

JDFPV, 2018-04-10:
Complete

JDFPV, 2018-04-10:
Sorry. i just seen where you said not to send in mutiple comments. For some reason using my phone it would not allow me to send my dump files in full.

biomedchad, 2018-04-10:
2:19 and 5:30.  Can’t see how it could be any clearer unless i blow up the text. Did you watch the video or skip around?  But I’ll get a log when weather cooperates.

shellixyz, 2018-04-10:
You should check your video link. I can only see one explanation. You linked the wrong video.

biomedchad, 2018-04-10:
https://www.youtube.com/watch?v=v13BZC4VSUc&app=desktop

biomedchad, 2018-04-10:
Dang my bad. Ok here it is. For real this time.

shellixyz, 2018-04-10:
@biomedchad I think your HDOP was just too high in the first flight. Seems better on the second flight when you have more sats. Also understand that the heading doesn't mean anything if you don't have a mag and are not moving. We will see with a log.

shellixyz, 2018-04-10:
@JDFPV I can't see anything wrong with your config. Are you sure your wing can climb at 20° with throttle at 1700 ? Otherwise it is possible it is just stalling. Did you use the same config than with 1.8 ? We need more info to draw any conclusion. Log and more details on what is happening.

biomedchad, 2018-04-10:
it seems like letting the gps warm up on the ground and gain more sats and improve hdop corrected my issue here after 1 flight.  more testing to come

closed 2018-06-24 by stale[bot]