iNavFlight/inav #3029: OSD CUTTING OUT, SATS VISIBLE, CONTROLS AND RTH STILL WORK

biomedchad, 2018-04-10:
### clracing f4air v2

inav 1.9
### Behavior
osd cut out on flight but sats remained and still had control..rth and functions still worked

VIDEO
https://youtu.be/2wF7qq-CWng

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/CLRACINGF4AIRV2 1.9.0 Mar  8 2018 / 06:50:40 (dc4ef594f)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1538 -55 -1 
servo 4 1000 2000 1472 55 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
[... 403 lines cut ...]
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 30
set tpa_breakpoint = 1600
set rc_expo = 50
set rc_yaw_expo = 20
set roll_rate = 28
set pitch_rate = 16
set yaw_rate = 20
set manual_rc_expo = 50
set manual_rc_yaw_expo = 20
set manual_roll_rate = 70
set manual_pitch_rate = 70
set manual_yaw_rate = 100

digitalentity, 2018-04-11:
Most likely hardware issue. Make sure you have power supply for your camera, VTX and flight controller well filtered and noise free.

shellixyz, 2018-04-11:
As digitalentity stated it looks like an electrical noise issue and your other [issue](https://github.com/iNavFlight/inav/issues/3021) is more than likely related. Noise in your GPS power supply can hinder its ability to receive satellites and its precision.

fiam, 2018-04-11:
Please, don’t use all caps in the issue titles.

As others have pointed out, this looks very much like a HW issue.

biomedchad, 2018-04-11:
Well those are fantastic theories but why would the same issue with the Gps occur on another ship with different components. And others have reported this issue. Are we all building noisy setups then?

Btw my esc and power lead are capped. My video is clean and grounded. I’ve been in rc and fpv for 6 years and know what noise looks like. I’m also pulling like 4 amps on a 30 amp esc. 

I’m willing to say maybe the osd chip could be bad but I would like to test some things out. Are you guys devs or just making comments here?  

Here is what i will do. Check my connections. Revert to basic font vs bold. Eliminate some info coming into the osd. And see where that gets me. 

Those sound like reasonable steps I think?

shellixyz, 2018-04-11:
> My video is clean and grounded

Why is your video recording looking so bad then ?

> And others have reported this issue

Links ?

> Are we all building noisy setups then?

I don't know but I never had any GPS nor OSD issues and it is often a noise issue when it happens or a problem with the OSD chip. Examples: #2913 #2419 

> Are you guys devs or just making comments here?

We are all 3 devs. Digitalentity is the original creator of the project and owner of the repos, fiam is a member of the project and great contributor and myself is an humble contributor just trying to help.

> Those sound like reasonable steps I think?

Sure can't hurt even though I don't think changing the font will solve anything.

krzysztofmatula, 2018-04-11:
Just out of my curiosity, why are you guys concluding it is a HW issue?
Unclean power could cause OSD flickering / chip resetting, but why the sat count is still on and working? I'd rather suspect some memory corruption...

fiam, 2018-04-11:
@biomedchad @krzysztofmatula I've seen several builds where the video ground was not properly grounded (e.g. an easy to miss one: VTX with multiple ground pins is left with one of them unconnected) and eventually some noise will confuse the max7456 which will just stop drawing either some areas or the whole screen until you power cycle it. It looks very similar to this video and was is usually a pain to diagnose and fix. 

It could definitely be a defective MAX7456. At this point I've seen them fail in basically any way you can imagine (I've spent quite a few hours doing OSD related stuff in INAV), so I wouldn't discard that yet.

Reasons that make me believe this is not a software issue:

- The FC is not hanging, otherwise the OSD would stop updating and it would keep the existing text (even blinking, since that's handled by the OSD chip itself).
- Overrunning a buffer would very likely bring down the aircraft shortly after, either by hanging the FC or either by corrupting some important data in RAM.
- The FC is still sending data to the max7456, since the satellite count is updated a few seconds later from 17 to 18.

As for possible ways to fix or mitigate the problem, I'd suggest:

- Make sure all the video system (camera, vtx and OSD) is properly grounded to a single place (e.g. don't ground camera to PDB and VTX to OSD GND).
- Turn on force_sw_blink. This will make the FC handle the blink instead of the MAX7456. There were some users which reported some areas of the OSD disappearing when hardware based blinking was enabled, so we added the option to make it software based and I know for sure it fixed the problem for at least one of them.
- Reduce the noise that reaches the MAX7456 as much as you can. I guess most of the noise we see in that video comes from poor reception, since the model was far at that point. Could you post some video when it was closer so we can see how the electrical noise was?

badandy1223, 2018-04-12:
I would not fly the ship until this is diagnosed properly. I had a Mini Drak with the same symptoms. What ended up happening is I lost all control of the aircraft. Motor cut, and the aircraft stalled and crashed 1 mile away from SP. I haven't recovered it, so I can't tell you exactly what happened. Just some friendly advice.

closed 2018-04-18 by biomedchad