iNavFlight/inav #3034: Drone bouncing gps mode after moving

danylam, 2018-04-11:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

danylam, 2018-04-11:
Hi, I've been having problem getting poshold + althold. When in this mode it stays steady but once I move the stick forward and then after 5 meter I leave the sticks in center, it starts to move backwards then forward. Bouncing... I think it has something to do with mag, but can't find a solution. Tried many configuration... No luck. Hope you can help me.
Thanks
Entering CLI Mode, type 'exit' to return, or 'help' # DUMP

# version
# INAV/KAKUTEF4V2 1.9.0 Mar 5 2018 / 01:18:34 (dc4ef594f)

# resources

# mixer
mixer QUADX

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature GPS
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 395 lines cut ...]
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
[blackbox_log_2018-04-11_162553.TXT](https://github.com/iNavFlight/inav/files/1900415/blackbox_log_2018-04-11_162553.TXT)
[blackbox_log_2018-04-11_162553.TXT](https://github.com/iNavFlight/inav/files/1900420/blackbox_log_2018-04-11_162553.TXT)

matlj, 2018-04-12:
I've had the exact same issue on my 330 after upgrading to 1.9.0, without changing any setting, and so do some people on rc-groups (violent stop and bounces when in poshold mode). 
Setting nav_mc_vel_xy_p to 60 fixed the issue for me.

danylam, 2018-04-12:
Thanks because it fixed for me as well

matlj, 2018-04-13:
Cool! 

This is really weird, because the same quad on 1.8.0 with default values for nav_mc_vel stopped very smoothly in POSHOLD mode. I found a dump from my quad on 1.8.0, and the value for p was even higher:
set nav_mc_vel_xy_p = 180

alexeystn, 2018-04-13:
My 5" quad also oscillates severely in Poshold with 1.9 firmware and default 5"GPS preset. Meanwhile 1.8 is almost perfect.
Here is what I found in code:
https://github.com/iNavFlight/inav/blob/1.9.0/src/main/flight/pid.c#L129
https://github.com/iNavFlight/inav/blob/1.8/src/main/flight/pid.c#L129
There is different scale for NAV_VEL_XY_P in versions 1.8 and 1.9 (×20 and ×100), so these values cannot be simply transfered from 1.8 to 1.9 with "diff" command.
I suppose, this is the reason.

digitalentity, 2018-04-13:
@alexeystn oops, good point. I've amended the release notes.

digitalentity, 2018-04-13:
@DzikuVx @fiam we should probably change presets to reflect the change

DzikuVx, 2018-04-13:
I will take care about it

svhirst, 2018-05-11:
I have found I had to put the D to the max 255 to control the issue on 1.9.3, admittedly I'm using an old quad.

closed 2018-07-24 by stale[bot]