iNavFlight/inav #3035: Bluetooth Module Not Working with CL_Racing F4 Air V2 Board

Zangetsu57, 2018-04-12:
I'm using a CL_Racing F4 Air V2 board with a bluetooth (BT) module connected to one of the UARTs.  I'm having issues connecting to the configurator via the BT module. The same BT module works fine with another omnibus board I have in another plane (also running iNav 1.9) so I know it is configured and binded correctly to my pc. I've verified that Tx is connected to Rx and vice-versa.  Additionally, I tried a separate BT module and am getting the same results.

JyeSmith, 2018-04-12:
When hooked up to a to a uart that is not uart1 (this is for sbus on the Air) you need to switch on MSP in the Ports tab.

Zangetsu57, 2018-04-14:
I did turn on MSP.

closed 2018-04-16 by DzikuVx