iNavFlight/inav #3044: INav 1.9 MatekF405 CTR CrossFire Failsafe not working

jalao, 2018-04-14:
I am currently running Matek 405 CTR using Crossfire Micro. I configured my crossfire failsafe to Cut and FS Pos and tested failsafe in both modes. I entered INAV and set failsafe to RTH. regardless of which option i choose, the moment I switch off my transmitter, the motors cut. I have also logged into CLI and made sure my parameters are set to RTH. Is this an issue or is there a workaround. 

THanks

### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

jalao, 2018-04-18:
Hi. Just checking in to see if you guys have been able to into my data dump to see what’s wrong.

jalao, 2018-04-30:
Hello.
Can you please provide an update. I lost a drone today because failsafe did not trigger RTH while using crossfire. Can you please, please help....

Thanks

solmag, 2018-05-01:
Hello, I am having the same problem with my crossfire setup. I know that I set it up correctly on my Inav but can somebody please explain how to set it up on crossfire? Maybe I'm doing something wrong. FYI I have set it up on Frsky L9R and it works perfectly. But I'm finding it extremely difficult to make FS RTH work on crossfire. Each time I shut off my transmitter, the motors shut off after about 2 seconds. Any help will be appreciated. Thanks

markandkymward, 2018-05-26:
One work around is if you have an available channel to broadcast RSSI on your transmitter and use this channel to trigger FAILSAFE and then RTH.  

Steps:
1. Configure LQ to output on your desired channel.  Do this by selecting MICRO or DIVERSITY (depending on the transmitter module) and then select OUTPUT MAP.  Set Dst Chan # to a free channel.
2. In INAV:
  a. Go to RECEIVER tab and set RSSI Channel to the channel you selected above.  Check that you are good by power cycling your transmitter.  You should see that channel drop from ~2000 to ~1000 on the channel you selected.
  b.  Go to the MODES tab and set the FAILSAFE mode to that same channel and put the range at ~900 to ~1300.  I'm not sure where 70% LQ lies on this spectrum but when you hit that LQ, you pretty much need to be turning back.  The 1300 is just a guess, the real number is probably higher than that.  This method will definitely handle a power loss on your transmitter though.  With the crazy range of the XF, this will likely be OK.  If you are planning any hardcore distance flying, you might want to experiment with this.
  c.  Go to the FAILSAFE tab and configure RTH as the default FAILSAFE mode.
  d.  Finally, go to the ADVANCED TUNING tab and configure your RTH details.

This method works well on both my XF diversity and my micro builds.

Hope this helps

closed 2018-08-09 by stale[bot]