iNavFlight/inav #3045: MAVLINK protocol over softserial

blackmennewstyle, 2018-04-14:
I'm trying to use MAVLINK protocol over softserial on the Omnibus F4 Pro from Airbot using the proper informations provided by the wiki for this board.

It's running iNAV 1.9.0 firmware. MAVLink protocol on softserial.

It seems to work, i'm getting correct datas from GPS, altitude, heading but i'm getting fluctuating values for VBAT & CURRENT, even though CURRENT sometimes is accurate.

Is there a recommendation about using MAVLINK over softserial?

digitalentity, 2018-04-14:
Do you get correct readings of VBAT and CURRENT in Configurator?

blackmennewstyle, 2018-04-14:
Yes, it's perfect in the GUI and OSD.
But it's not accurate over the MavLink Telemetry.

stronnag, 2018-04-14:
I bench tested iNav's mavlink over a hardware UART (in order to remove any deficiencies in the transport from the test; SPRF3, hardware UART / 57600 baud). Voltage (and all other data) is completely stable.

Ran the test for 600s (10min), receiving 21985 Mavlink messages at 1051 bytes/sec. 0 CRC errors.

It appears that iNav's Mavlink is stable, where the data link can cope and the receiving application behaves appropriately. Whether soft serial is happy sustaining 1051 b/s is another matter.

blackmennewstyle, 2018-04-15:
Interesting feedbacks.

I'm currently running 19200 bauds. I will try to slow it down and let you know the results.

blackmennewstyle, 2018-04-15:
I just tried with 9600 bauds and the problem remains the same.

The datas are all over the place and inaccurate when the quad is armed.

As soon as i disarmed, the datas are good but and display the exact values with the GUI at this exact moment.

blackmennewstyle, 2018-04-15:
By the way in the CLI, when i perform set telemetry_switch=ON and save.
I stop receiving telemetry whatever the quad is armed or not.

stronnag, 2018-04-15:
On a SPRF3, Mavlink / SS will not work at 9600, it cannot sustain the rate (1050 bytes/sec).
At 9600, I see at around 5% CRC error, presumably due to data overruns.
```
2018-04-15T11:40:09+0100 Comm error count 244
2018-04-15T11:40:09+0100 MAVCRC Fail, got 0 != 1c4 (cmd=74, len=20)
2018-04-15T11:40:10+0100 186s, rx 150045b, tx 306b, (805b/s, 2b/s) messages 4978
``` 
With softserial at 19200 baud, mavlink is stable and sustainable, no errors. Consistent reception of data.

blackmennewstyle, 2018-04-15:
Thanks for the feedbacks!

The value for VBAT is innacurate whatever the baudrate that i use :'(

It's possibly an issue related to my UHF system. I'm using the DragonLink V3 which supports MAVLink protocol. Maybe i need to to dig more in that way.

Thanks for your time anyway.

stronnag, 2018-04-15:
I don't think this is an iNav issue; with a direct connection,  VBAT is consistent even when I'm dropping frames at 9600 baud.

digitalentity, 2018-04-15:
Dragonlink is probably not a transparent link - it probably does MAVLink decoding at air side, transmits over the air using it's own packed protocol and reconstructs MAVLink on ground side (that would I would do if I would develop an LRS).

blackmennewstyle, 2018-04-15:
I got it working guys :)

Indeed problem was in my DragonLink settings configuration. My bidir bandwith was way too slow and my RF update preference also, i set everything to fastest update rate. And now my values are accurate and consistent.

Thanks for the help. You can close this issue :)

closed 2018-04-15 by digitalentity

pitersonfascist, 2020-02-07:
I found the reason and solution for it as well!
Mavlink won't work if your Softserial's RX and TX are configured on the same resource.

I looked into the source code of Mavlink, and it opens a bidirectional connection, which means that the same resource pin works as input and output at the same time.

I changed the configuration in my `target.h` to have different resources for my Softserial and Mavlink started working well on my Matek-F11 WSE on TX2 pin

WooHoo, now I'll have telemetry on my Qczek LRS =)

gshanbhag525, 2020-03-11:
hi , i wanted to know if i can connect raspberry pi to omnibus f4 pro using mvalink??

pitersonfascist, 2020-03-11:
Sure you can, but not using iNav. 

iNav can only send telemetry via MavLink, but Ardupilot can do it bidirectional.

pitersonfascist, 2020-08-24:
You need the source code of this repo.
Locate your target  MATEKF411SE and in the `target.h`
#define USE_SOFTSERIAL2
 #define SOFTSERIAL_2_TX_PIN     PA2 // TX2 pad
-#define SOFTSERIAL_2_RX_PIN     PA2
+#define SOFTSERIAL_2_RX_PIN     PB10

made respective changes, so replace SOFTSERIAL_2_RX_PIN with LED pin

Build and flash a new firmware.

Igor-B-v, 2021-03-24:
Ievgenii
can you share your hex file for inav 2.6.1?

pitersonfascist, 2021-03-26:
> Ievgenii
> can you share your hex file for inav 2.6.1?
Here is, but I didn't try it yet, still flying on 2.5
[inav_2.7.0_MATEKF411SE.tar.gz](https://github.com/iNavFlight/inav/files/6212959/inav_2.7.0_MATEKF411SE.tar.gz)

vihmane, 2021-05-13:
> > Ievgenii
> > can you share your hex file for inav 2.6.1?
> > Here is, but I didn't try it yet, still flying on 2.5
> > [inav_2.7.0_MATEKF411SE.tar.gz](https://github.com/iNavFlight/inav/files/6212959/inav_2.7.0_MATEKF411SE.tar.gz)

I can't find a configurator for 2.7, and at what speed does the port work?

pitersonfascist, 2021-05-13:
That's what I built from the master branch, it should be 2.6.

For mavlink, as it is a softserial, as low as possible.

9600, 38400. Definitely not a 115200

vihmane, 2021-05-13:
And my 2.6.1 comes with an error: This firmware version is not supported. This version of Configurator supports firmware from 2.5.0 to 2.7.0 (excluded)

vihmane, 2021-05-13:
> That's what I built from the master branch, it should be 2.6.
> 
> For mavlink, as it is a softserial, as low as possible.
> 
> 9600, 38400. Definitely not a 115200

And my 2.6.1 comes with an error: This firmware version is not supported. This version of Configurator supports firmware from 2.5.0 to 2.7.0 (excluded)
your firmware is 2.7.0

vihmane, 2021-05-14:
i found another solution. connection to s1. Everything is working. I'm happy http://www.mateksys.com/?p=5159#tab-id-8

pitersonfascist, 2021-05-17:
Yes, the firmware version MATEKF411SE_S6_SR1 completely solves this topic

pitersonfascist, 2021-05-30:
[inav_2.6.1_MATEKF411SE.tar.gz](https://github.com/iNavFlight/inav/files/6565807/inav_2.6.1_MATEKF411SE.tar.gz)

alexvazquez, 2024-10-29:
@pitersonfascist  hi, I haven't connected my FC to INAV for 3 years and now is forcing me to update firmware and I have it the MATEKF411SE_S6_SR1  installed right now. Im using MatekF411 SE with Caddx Vista.

There are the firmwares that INAV shows me and dont know which one to choose

[image]

pitersonfascist, 2024-10-29:
[alexvazquez](https://github.com/alexvazquez) Please have a look at the product page from the post above 
[http://www.mateksys.com?p=5159#tab-id-8](http://www.mateksys.com/?p=5159#tab-id-8)

> Variant:
MATEKF411SE Stock target. LED control and have SS1 on ST1 pad, SS2 on TX2 pad.
MATEKF411SE_PINIO Adds USER 2 PINIO support on the LED pad.
MATEKF411SE_FD_SFTSRL1 Adds full duplex SS1 by putting the RX on the LED pad.
MATEKF411SE_SS2_CH6 SS2 moved to Ch6 pad. This keeps UART 2 as a full UART (for example, for use with Crossfire) and SS2 support.

it looks that you need MATEKF411SE_SS2_CH6

alexvazquez, 2024-10-29:
Appreciate sooooooo much!!!

 
 

http://www.mateksys.com?p=5159#tab-id-8